EmbeddedRPC/erpc #104: ThreadX OS support

tmordeko, 2020-07-01:
Hi,

One of my cores is working with ThreadX OS, is there support for this OS in the porting layer ? from some reason I couldn't find explicit reference to ThreadX.

/Tzvika

Hadatko, 2020-07-01:
Hi @tmordeko , here are porting files for C projects https://github.com/EmbeddedRPC/erpc/tree/develop/erpc_c/port
ThreadX OS is not supported yet. Feel free to do implementation by yourself and then you can add it to official repository trough pull request. Implement changes on devel branch. You can take inspiration from other ports.

Last time mbed support was added through pull request:
https://github.com/EmbeddedRPC/erpc/pull/81/files
this is wating for merge:
https://github.com/EmbeddedRPC/erpc/pull/92/files

tmordeko, 2020-07-01:
Thanks @Hadatko , is there plan to support it in the near future ?

Hadatko, 2020-07-01:
Don't know about any future plans with this OS. @MichalPrincNXP

MichalPrincNXP, 2020-07-14:
Hi, there is no plan to support ThreadX OS, unless sb. makes the support and creates the pull request like with #81 .

Hadatko, 2022-01-03:
Done https://github.com/EmbeddedRPC/erpc/pull/175

closed 2022-01-03 by Hadatko