EmbeddedRPC/erpc #105: rpmsg transport

tmordeko, 2020-07-12:
Hi,

Appreciate if someone could elaborate on rpmsg transport, the documentation is lacking information on this topic.
What is it used for in erpc context ? what is the difference between all existing flavors ? could it work over uart ? point to any example of usage could assist as well.

Thanks in advance,
Tzvika

Hadatko, 2020-07-12:
Hi @tmordeko ,
you can find more information about rpmsg here: https://github.com/NXPmicro/rpmsg-lite
Example: https://github.com/EmbeddedRPC/erpc-imx-demos

tmordeko, 2020-07-14:
Hi @Hadatko 
Thanks for quick response
Assuming I have no shared memory between cores, only uart interface. and I would like to use rpmsg mechanism is it doable ? can you point me to such example or guide me ?

/Tzvika

MichalPrincNXP, 2020-07-14:
Hi @tmordeko , rpmsg_lite transport layer is intended to be used just in cases when erpc runs on multicore platforms (dual-core MCU) leveraging shared memory for data exchange. RPMsg_Lite does not work with UART.

closed 2022-05-03 by Hadatko