EmbeddedRPC/erpc #106: endianness agnostic

tmordeko, 2020-07-14:
Hi,

Is it possible to run erpc between two endian different cores ? do you plan to support it ?

/Tzvika

MichalPrincNXP, 2020-07-14:
Hi @tmordeko , this is not supported. Will add this requirement into the TODO list. If you would be able to implement this and prepare the pull request we would appreciate it. Thank you.

tmordeko, 2020-07-14:
Hi @MichalPrincNXP , we will try to do that although not sure we have all the familiarity needed from erpc library, we will need some guidance here as it involves sensitive and complicated areas , anyway we can assist in testings if we get evaluation code from you if that helps. Thanks

MichalPrincNXP, 2020-07-15:
Hi @tmordeko , could you describe your use case a little? What cores and the transport layer are you using? Thanks.

Hadatko, 2020-07-15:
Hi, it looks like good place to do it for C is in https://github.com/EmbeddedRPC/erpc/blob/develop/erpc_c/infra/erpc_message_buffer.cpp

Cursor read/write function... Didn't analysed that in deep...

tmordeko, 2020-07-16:
Hi @MichalPrincNXP , @Hadatko 

One cpu is little endian while the other is big , the transport layer is UART (in the future considering to move into rpmsg) and I am using almost all IDL data types
I did some changes more like workarounds in both sides and it's working fine,  in:
void BasicCodec::startReadMessage(message_type_t *type, uint32_t *service, uint32_t *request, uint32_t *sequence)
void BasicCodec::startReadUnion(int32_t *discriminator)
void BasicCodec::readBinary(uint32_t *length, uint8_t **value)
erpc_status_t FramedTransport::receive(MessageBuffer *message)
But I am sure there is more general place where  the fix can be placed like @Hadatko mentioned,   I am sure more users can benefit from it :-)

/Tzvika

MichalPrincNXP, 2020-07-22:
Hello @tmordeko ,
I think @Hadatko is right, could you please try to put your endianness conversion code into 
MessageBuffer::Cursor::read
MessageBuffer::Cursor::write
and try it with your use case, please? Thanks.

MichalPrincNXP, 2020-07-22:
Well, changes in 
MessageBuffer::Cursor::read
MessageBuffer::Cursor::write
will ensure streamed data conversion but we still also need to convert message header (BasicCodec::startReadMessage)

tmordeko, 2020-07-22:
Hi @MichalPrincNXP 

when you say changes in:
MessageBuffer::Cursor::read
MessageBuffer::Cursor::write

as far as I understand the msg header reading is also using MessageBuffer::Cursor::read ? the only operation there is memcpy which is endian agnostic since it is byte wise, how do you recommend to modify these functions such that will be able to handle all data types (IDL types) and parse erpc header ?

/Tzvika

MichalPrincNXP, 2020-07-27:
Hello @tmordeko , yes, you are right that the header is read using MessageBuffer::Cursor::read, I have missed that.
Anyway, there are two possible ways how to handle endianness, I think. The first one is the approach you started with, i.e. making endianness conversion for each supported data types (in BasicCodec::startReadUnion, BasicCodec::readBinary,... functions). The second one is to make the endianness conversion in more general function (MessageBuffer::Cursor::read as @Hadatko recommends). The second approach can be used for standard data types with 2/4/8 bytes but it will not work with lists, binaries, etc. It could be solved in a way that 2/4/8 bytes long data types will be converted in MessageBuffer::Cursor::read (checking the length parameter) and the rest of "non-standard" data types will be converted in upper layer functions like BasicCodec::startReadList. @Hadatko , could you please comment? Thanks.

tmordeko, 2020-09-30:
Hi @MichalPrincNXP , @Hadatko 

I didn't see any comment, is there a plan to add this support to next release as part of the enhancement request ? I think a lot of users can benefit from it in the future :-)

/Tzvika

Hadatko, 2020-09-30:
Well if we are waitning for my comments it will have to wait. I am very busy these days.

MichalPrincNXP, 2020-09-30:
I am sorry, I am not able to address this issue in the next planned release ... will have to postpone.

amgross, 2022-05-17:
It seems to me that adding it to Cursor::read/write depending on the length won't work.
For example: string of 2 bytes will look like uint16_t, and float will look like uint32_t (according the web, float and double need carefully care according each architecture, as there is no standards)

PR #276 (2022-06-07): Add defines for changing transport to endianness agnostic
# Pull request

**Choose Correct**

- [ ] bug
- [X] feature

**Describe the pull request**

Added defines for user for changing the bytes that transport to some endianness agnostic way he decide on.

**To Reproduce**

**Expected behavior**

**Screenshots**

**Desktop (please complete the following information):**

- OS:Windows
- eRPC Version:1.9.0

**Steps you didn't forgot to do**

- [X] I checked if other PR isn't solving this issue.
- [X] I read Contribution details and did appropriate actions.
- [X] PR code is tested.
- [X] PR code  [... cut]