EmbeddedRPC/erpc #107: Client timeout and multiple client support using same transport.

anirudhgargi, 2020-07-15:
Hello, 

I had 2 queries for my implementations. 

1) **Client Timeout**: Currently client will get stuck in infinite wait in case something is wrong during the eRPC. Is there a way we can a timeout for each of the eRPC call ? Idea is to let client execute ahead and be aware some error has happened. 

2) **Multiple client**: Is there a simple way which there can be more than 1 clients talking to the same eRPC server over the same transport. I was reading about the transport arbitrator, is it possible to use that arbitrator among multiple clients ? 
Right now its a arbitrator between client and server. Right ? 

My use case looks something like this : 

OS 1     |               |    OS2
=====|======|======
Server--|--UART--|---client1
             |               |---client2                     

Thanks.

anirudhgargi, 2020-07-21:
@Hadatko : Can you suggest something for query about multiple clients using same server. 
From a comment made on this forum, i happen to read that there shall be only 1 erpc Server and 1 erpc client using the transport. Application logic on the client side need to figure out a way to serialize and distinguish the clients, erpc will not do it.  Is this understanding correct ?

anirudhgargi, 2020-07-27:
@MichalPrincNXP : Can you share some thought on these 2 ? Thanks.

MichalPrincNXP, 2020-07-27:
Hello @anirudhgargi , sorry for delayed response. Regarding your question:
1. There is no possibility to define timeout for each eRPC request, it is up to the transport layer to implement a timeout and return kErpcStatus_ReceiveFailed when no response obtained within a timeout (in case of UART transport this could be implemented in UartTransport::underlyingReceive if needed).
2. As discussed in #57 having two clients on one side is not supported yet. Your statement that "application logic on the client side need to figure out a way to serialize and distinguish the clients" is correct.

closed 2020-08-04 by anirudhgargi