EmbeddedRPC/erpc #109: Is there a Todo list or roadmap for the software?

Pillar1989, 2020-07-24:
@MichalPrincNXP What is the future of the software?

MichalPrincNXP, 2020-07-27:
Hello @Pillar1989 , the erpc project development continues, the goal is to provide at least two releases per year (aligned with NXP MCUXpressoSDK releases). The list of tasks, todos, bugs is maintained internally. We are trying to address this Github community issues and pull requests first. Except of bugfixing and accepting new community pull request, new transports are planned for the next release. Hope it helps.

Pillar1989, 2020-07-28:
Thank you for your answer. When is the next release? What feature is new transports? Can you divulge?  @MichalPrincNXP

MichalPrincNXP, 2020-07-28:
The next release is planned in 6 months (beg. of January). New transports means we are going to add support for new transport layers: except of RPMsg_Lite, UART, SPI, serial, TCP, MU - based transports (currently available) there is a plan to support transports based on Linux SPIdev and I2C.

closed 2020-08-02 by Pillar1989