EmbeddedRPC/erpc #112: Errors using erpc with Zephyr C++

jpdoyon, 2020-08-05:
What's the level of maturity of erpc 1.7.3 with Zephyr?  I'm getting many errors when trying to compile with Zephyr 2.2.99 (see below).

`FAILED: CMakeFiles/app.dir/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp.obj 
ccache /usr/zephyr-sdk-0.11.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-g++  -DBUILD_VERSION=zephyr-v2.2.0-1903-gcd795c1b3bc8 -DKERNEL -DNRF52833_XXAA -D_FORTIFY_SOURCE=2 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/kubuntu/npdst/app/src/erpc/erpc_c/config -I/home/kubuntu/npdst/app/src/erpc/erpc_c/infra -I/home/kubuntu/npdst/app/src/erpc/erpc_c/port -I/home/kubuntu/npdst/app/src/erpc/erpc_c/setup -I/home/kubuntu/npdst/app/src/erpc/erpc_c/transports -I/home/kubuntu/zephyr/include -Izephyr/include/generated -I/home/kubuntu/zephyr/soc/arm/nordic_nrf/nrf52 -I/home/kubuntu/zephyr/lib/libc/newlib/include -I/home/kubuntu/modules/hal/cmsis/CMSIS/Core/Include -I/home/kubuntu/modules/hal/nordic/nrfx -I/home/kubuntu/modules/hal/nordic/nrfx/drivers/include -I/home/kubuntu/modules/hal/nordic/nrfx/mdk -I/home/kubuntu/modules/hal/nordic/. -I/home/kubuntu/modules/debug/segger/rtt -I/home/kubuntu/npdst/app/drivers/sx1276/zephyr/. -Os -fcheck-new -std=c++11 -Wno-register -fno-exceptions -fno-rtti -imacros/home/kubuntu/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -imacros/home/kubuntu/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-address-of-packed-member -Wpointer-arith -Wno-unused-but-set-variable -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/kubuntu/npdst/app=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/kubuntu/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/kubuntu=WEST_TOPDIR -ffunction-sections -fdata-sections -specs=nano.specs -MD -MT CMakeFiles/app.dir/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp.obj -MF CMakeFiles/app.dir/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp.obj.d -o CMakeFiles/app.dir/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp.obj -c /home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp
/home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp: In constructor 'erpc::Thread::Thread(const char*)':
/home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp:27:12: error: no matching function for call to 'k_thread::k_thread(int)'
   27 | , m_stack(0)
      |            ^
In file included from /home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading.h:26,
                 from /home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp:9:
/home/kubuntu/zephyr/include/kernel.h:537:8: note: candidate: 'k_thread::k_thread()'
  537 | struct k_thread {
      |        ^~~~~~~~
/home/kubuntu/zephyr/include/kernel.h:537:8: note:   candidate expects 0 arguments, 1 provided
/home/kubuntu/zephyr/include/kernel.h:537:8: note: candidate: 'constexpr k_thread::k_thread(const k_thread&)'
/home/kubuntu/zephyr/include/kernel.h:537:8: note:   no known conversion for argument 1 from 'int' to 'const k_thread&'
/home/kubuntu/zephyr/include/kernel.h:537:8: note: candidate: 'constexpr k_thread::k_thread(k_thread&&)'
/home/kubuntu/zephyr/include/kernel.h:537:8: note:   no known conversion for argument 1 from 'int' to 'k_thread&&'
/home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp: In constructor 'erpc::Thread::Thread(thread_entry_t, uint32_t, uint32_t, const char*)':
/home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp:38:12: error: no matching function for call to 'k_thread::k_thread(int)'
   38 | , m_stack(0)
      |            ^
In file included from /home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading.h:26,
                 from /home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp:9:
/home/kubuntu/zephyr/include/kernel.h:537:8: note: candidate: 'k_thread::k_thread()'
  537 | struct k_thread {
      |        ^~~~~~~~
/home/kubuntu/zephyr/include/kernel.h:537:8: note:   candidate expects 0 arguments, 1 provided
/home/kubuntu/zephyr/include/kernel.h:537:8: note: candidate: 'constexpr k_thread::k_thread(const k_thread&)'
/home/kubuntu/zephyr/include/kernel.h:537:8: note:   no known conversion for argument 1 from 'int' to 'const k_thread&'
/home/kubuntu/zephyr/include/kernel.h:537:8: note: candidate: 'constexpr k_thread::k_thread(k_thread&&)'
/home/kubuntu/zephyr/include/kernel.h:537:8: note:   no known conversion for argument 1 from 'int' to 'k_thread&&'
/home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp: In member function 'void erpc::Thread::start(void*)':
/home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp:56:118: error: invalid conversion from 'void* (*)(void*, void*, void*)' to 'k_thread_entry_t' {aka 'void (*)(void*, void*, void*)'} [-fpermissive]
   56 |     k_thread_create(&m_thread, m_stack, m_stackSize, threadEntryPointStub, this, NULL, NULL, m_priority, 0, K_NO_WAIT);
      |                                                                                                                      ^
      |                                                                                                                      |
      |                                                                                                                      void* (*)(void*, void*, void*)
In file included from /home/kubuntu/zephyr/include/kernel.h:5401,
                 from /home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading.h:26,
                 from /home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp:9:
zephyr/include/generated/syscalls/kernel.h:66:131: note:   initializing argument 4 of 'k_thread* k_thread_create(k_thread*, k_thread_stack_t*, size_t, k_thread_entry_t, void*, void*, void*, int, u32_t, k_timeout_t)'
   66 | static inline k_tid_t k_thread_create(struct k_thread * new_thread, k_thread_stack_t * stack, size_t stack_size, k_thread_entry_t entry, void * p1, void * p2, void * p3, int prio, u32_t options, k_timeout_t delay)
      |                                                                                                                  ~~~~~~~~~~~~~~~~~^~~~~
/home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp: In member function 'bool erpc::Thread::operator==(erpc::Thread&)':
/home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp:61:21: error: no match for 'operator==' (operand types are 'k_thread' and 'k_thread')
   61 |     return m_thread == o.m_thread;
      |            ~~~~~~~~ ^~ ~~~~~~~~~~
      |            |             |
      |            k_thread      k_thread
/home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp: In constructor 'erpc::Mutex::Mutex()':
/home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp:94:12: error: no matching function for call to 'k_mutex::k_mutex(int)'
   94 | : m_mutex(0)
      |            ^
In file included from /home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading.h:26,
                 from /home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp:9:
/home/kubuntu/zephyr/include/kernel.h:3535:8: note: candidate: 'k_mutex::k_mutex()'
 3535 | struct k_mutex {
      |        ^~~~~~~
/home/kubuntu/zephyr/include/kernel.h:3535:8: note:   candidate expects 0 arguments, 1 provided
/home/kubuntu/zephyr/include/kernel.h:3535:8: note: candidate: 'constexpr k_mutex::k_mutex(const k_mutex&)'
/home/kubuntu/zephyr/include/kernel.h:3535:8: note:   no known conversion for argument 1 from 'int' to 'const k_mutex&'
/home/kubuntu/zephyr/include/kernel.h:3535:8: note: candidate: 'constexpr k_mutex::k_mutex(k_mutex&&)'
/home/kubuntu/zephyr/include/kernel.h:3535:8: note:   no known conversion for argument 1 from 'int' to 'k_mutex&&'
/home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp: In constructor 'erpc::Semaphore::Semaphore(int)':
/home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp:118:10: error: no matching function for call to 'k_sem::k_sem(int)'
  118 | : m_sem(0)
      |          ^
In file included from /home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading.h:26,
                 from /home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp:9:
[... 6 lines cut ...]
[... cut ...]
/home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp: In static member function 'static void* erpc::Thread::threadEntryPointStub(void*, void*, void*)':
/home/kubuntu/npdst/app/src/erpc/erpc_c/port/erpc_threading_zephyr.cpp:90:19: warning: control reaches end of non-void function [-Wreturn-type]
   90 |     k_thread_abort(k_current_get());
      |     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~
[11/130] Building CXX object CMakeFiles/app.dir/src/erpc/erpc_c/setup/erpc_server_setup.cpp.obj
`

Hadatko, 2020-08-05:
Hi @jpdoyon ,
somebody was updating zephyr related code, but as long as i know he never did pull request. 
You can try look here: https://github.com/masz-nordic/erpc/commits/zephyr_integration

Hadatko, 2020-08-05:
Hi @jpdoyon ,
maybe you can find fix in forks like:
https://github.com/bneumann/erpc/commits/zephyr_threading

jpdoyon, 2020-08-05:
Haven't tried running it yet, but yes, it does look much better.  Thank you so much!

closed 2020-08-05 by jpdoyon