EmbeddedRPC/erpc #114: erpc_mbf_dynamic_init not available to clients

dpfrey, 2020-08-10:
The function `erpc_mbf_dynamic_init` is used in the example on the wiki page: https://github.com/EmbeddedRPC/erpc/wiki/Getting-Started

I found that this function is declared in `setup/erpc_mbf_setup.h` and defined in `setup/erpc_setup_mbf_dynamic.cpp`, but the header isn't listed in the `HEADERS` section of the Makefile and the cpp file isn't listed in the `SOURCES` section of the Makefile.

Perhaps I am misunderstanding the way that eRPC is supposed to be used.  Is the user supposed to pick and choose cpp and h files that may not even be refererenced by the Makefile?

RamseyTI, 2020-08-12:
I'm no expert, but in general, you edit the source list to build only the files needed by your application. For example, you would only build the transport source files you intend to use. I can see how the example makefile would become out-of-date as more transports are added.

MichalPrincNXP, 2020-09-14:
Hello @dpfrey , thank you for this question. @RamseyTI is right, one should pick source files based on the application needs. This depends on the system/OS and the used transport layer. Anyway, the erpc_c\Makefile is intended to be used for Linux env. and some files that are relevant to Linux are missing there, incl. erpc_mbf_setup.h and erpc_setup_mbf_[dynamic, static].cpp sources. I will update it the next release.

Hadatko, 2022-05-03:
@MichalPrincNXP done?

MichalPrincNXP, 2022-05-04:
Yes, updated in https://github.com/EmbeddedRPC/erpc/commit/2526606a280f7b2d26d7adbed373e5c573dd0159

closed 2022-05-04 by MichalPrincNXP