EmbeddedRPC/erpc #122: callback question (function to call is on client, but server calls)

pgu-swir, 2020-09-14:
Hi - I'm struggling a bit with the following use case for callbacks: (just for context, I have bi-directional RPC working well, there is a server and a client on both side that communicate correctly)

- My client uses a "register" RPC that expects a pointer to a function to call back later. That works nicely then the function to call is on the server. The erpcgen generates a client shim for that function and passes it to the "register" RPC call and things happens as expected. When any function of the server calls the registered function, the "real" function is called on the server.

- Now I want my "register" RPC to register a pointer to a "local" function on the client (it happens to be of the same prototype than the case above). The "local" function is declared in the server part of the client and of course has its own implementation. I'm not sure how to tell the IDL that this 'local' function is already part of the RPC server system provided by the client. eRPC always creates a client shim, although I have the implementation already. Do I need to name the "local" function in IDL with a different "fake" name so that eRPC generates a client shim for that "fake" and on the server side, the server implementation of that "fake" function calls the "real" function that would happen to be a RPC (my head just exploded)

Hadatko, 2020-09-14:
Hi @pgu-swir , basically for safety reason we are always creating shim code for callback functions. This way you can call function from client side and server side (as local function or erpc call). Maybe you will not use erpc call in this case but you can ;) Currently you cannot get rid of of generated code. In future we can create empty erpc implementation to reduce code size if you use specific annotation...

Hadatko, 2020-09-14:
More information e.g. here: https://github.com/EmbeddedRPC/erpc/issues/101
or here: https://github.com/EmbeddedRPC/erpc/issues/47

pgu-swir, 2020-09-14:
That helps, thanks - I'll continue to work on that, I have some challenges as well with the generated code not compiling if I don't set @group

BTW "my head exploded" was not a critic :-)

pgu-swir, 2020-09-15:
This is working great now, thanks again. I did not realize that the magic was in the use of @group (I was trying to have 2 separated files). One last question though: with the following example (it's a JerryScript port)
```
@include("jerryscript.h")
@include("jerryscript-ext/handler.h")
program jerry

@external enum jerry_init_flag_t { }
@external enum jerry_parse_opts_t { }
@external enum jerry_error_t { }
@external enum jerry_type_t { }

@external type jerry_char_t = uint8
@external type jerry_value_t = uint32
@external type jerry_size_t = uint32
@external type jerry_length_t = uint32
@external type size_t = uint32

@external jerry_external_handler_t (jerry_value_t function_obj,
                          jerry_value_t this_val,
                          list<jerry_value_t> args_p @length(args_count),	
                          jerry_length_t args_count) -> jerry_value_t
@group(api) 
interface JerryAPI {
	@external jerry_call_function (jerry_value_t func_obj_val, jerry_value_t this_val,
                                   list<jerry_value_t> args_p @length(args_count), jerry_size_t args_count) -> jerry_value_t							
	
        @include("jerry_local.h")
	@external jerry_external_handler_t jerryx_handler_print;
                                   
}

@group(local)
interface JerryLocal {
         jerry_external_handler_t print_handler;    
}
```
The function "jerryx_handler_print" is running together with JerryScript and the function "print_handler" runs on the client that uses the remote JerryScript engine and can be called from JerryScript. 

But I had to include a header that defines "print_handler" (I used the auto-generated header) otherwise it is missing for the callback array in jerry_api_server.cpp. I understand it could be a user-defined header, but as "print_handler" is not external, could  the 1st group's server header include 2nd group general header? or at least repeat non-external definitions of the 2nd group?

Or do I misuse IDL again?

Hadatko, 2020-09-15:
Hi @pgu-swir , sorry i will try look on your issue later currently i am busy. But quickly: If i remeber it correctly tak annotations passed to calleback type will be inherited by callback functions. So both jerryx_handler_print and print_handler are declared as external because @external jerry_external_handler_t (jerry_value_t function_obj,
                          jerry_value_t this_val,
                          list<jerry_value_t> args_p @length(args_count),	
                          jerry_length_t args_count) -> jerry_value_t

But as i said it is long time ago. If you remove this external for callback type does it help? If not i will try find more time for respond later.

pgu-swir, 2020-09-15:
Unfortunately no, but again I can solve the issue issue by forcing a @include(jerry_local.h) that will include "jerry_local.h" in the jerry_api_server.cpp, otherwise jerry_api_server.cpp does not compile. Because these "group2" functions are referenced bye group1_server.cpp, they could be in its header when they are not @external, or the "group2.h could be included automatically in group1_server.h

I was just curious why and wanted to know if all non external functions of group2 could be included in server/general header of group1 to automatically solve that. When external, user is supposed to  bring his @include(my_header) so it's normal - still, no bid deal

I need to work on ERPC_NESTED_CALLS now :-)

Hadatko, 2020-09-15:
Hi @pgu-swir , Maybe during weekend i will look on it closer. I know that i was trying to do it smart, these includes, but something wasn't entirely clear (some usecases maybe). I need to try that on my own computer. Could you write erpcgen version your are using? To be sure we are using same. 

Crossing fingers with nested calls. If you will managed that you will have worst behind you ;) I hope... ;)

pgu-swir, 2020-09-15:
I have compiled from "develop" branch and it says 1.7.4. Re nested calls, I'm almost there, I have this other question about multi-level nesting

closed 2020-09-18 by pgu-swir