EmbeddedRPC/erpc #123: can nested calls be multi-levels

pgu-swir, 2020-09-15:
Hi - I hope the title is self-explanatory. Some server functions, called already in a nested context, need to call themselves client RPC and so far it locks. Is this because nesting is only one level?

Hadatko, 2020-09-15:
Hi @pgu-swir , here you can find multilevel nested example: https://github.com/EmbeddedRPC/erpc/tree/develop/test/test_arbitrator

But you need be carefull with that.

pgu-swir, 2020-09-15:
Thanks, I'll carefully re-read the example. The one-layer on nested works fine, but the lock happens when client A sends a request/RPC call to server B. On server B, the function calls a server A RPC and so far that works. The issue happens when the RPC called in server A itself does a server B request/RPC call

pgu-swir, 2020-09-16:
All fixed now and the multi-level nested works like a charm. I changed a bit the TCP transport to disable Nagle's algorithm as I think we want every send() to be sent immediately.

Thank again - I'll commit the platform specific part in a few days

closed 2020-09-16 by pgu-swir