EmbeddedRPC/erpc #145: add debugs to erpc auto-generated files

tmordeko, 2020-10-20:
Hi,

This is probably request for enhancement.

Is it possible to add debug messages / some mechanism for stats collection from within the auto generated files that will be integrated and part of the general erpc debug/stat (in continue #144)

/Tzvika

Hadatko, 2020-10-20:
Sounds like you would like to add annotation into erpc file which will add debug prints into shim code. I think currently is no such a function.

tmordeko, 2020-10-26:
Hi @Hadatko ,

On server side I would like to catch and be able to monitor errors earlier in the flow in a specific rpc flow in addition I would like to collect statistics which is only possible from the implementation of the auto-generated files before the control is passed to the rpc handler implementation.

/Tzvika