EmbeddedRPC/erpc #149: Different function generation Python and C/C++

ierturk, 2020-11-10:
Hi,
The input parameter which is used in `@length(req_len)` won't be generated for python and this parameter won't pass into the python function. Is this expected behavior? If this not the case, is there is a workaround to fix it? 

```
IDL -> erpc_spiRecvStream(in uint32 req_len, out binary data @length(req_len) @max_length(data_len_max), out uint32 ret_len) -> bool
C:/C++ -> bool erpc_tcpRecvStream(uint32_t req_len, uint8_t * data, uint32_t * ret_len)
PY -> def erpc_tcpRecvStream(self, data, ret_len)
```

Thank in advance

MichalPrincNXP, 2020-11-11:
Hello, based on my limited knowledge of the python generator and expected python code, I think this should be correct. Looking into the test_length_py output the IDL parameter that defines the length of the binary parameter is not passed into the python function. It is up to the user to pass the binary object with the appropriate length in this case, I think. @Hadatko / @RamseyTI / @zuzy , could you confirm please? Thanks.

ierturk, 2020-11-12:
Hi @MichalPrincNXP,
Thanks for the reply. If this is the case, it needs to be insert a duplicate parameter to pass `length` into Python side. In my case client was implemented in C/C++ and server was Python. So this may cause some complication without knowing binary data length in C/C++ when returned binary array size.
Thanks and regards.

ierturk, 2020-11-27:
Even though the function parameter in `@max_length(value)` is defined as `inout`, while the parameter won't generated in `Python `version, but it is generted in `C/C++` version.

MichalPrincNXP, 2020-12-16:
I am sorry not being able to provide an exact comment and response. I think even inout direction is used the behavior on the python side is ok. Anyway, please use another parameter to pass the length for this special C/C++ vs. Python use case.

Hadatko, 2022-05-08:
Some features are not implemented on python side as C language was mainly developed one.

closed 2023-11-30 by ierturk