EmbeddedRPC/erpc #150: Multiple erpc requests processing (server side)

tmordeko, 2020-11-15:
Hi,

Assuming 2 cores (A+B) communicating over serial interface, like UART, A core acting as client and B as rpc server.

Is it possible to process multiple requests (on the server side) coming from two different threads A.1 and A.2 simultaneously ?

A.1 ----first erpc msg  ---> B
context switch A
A.2 ----first erpc msg ---> B

How B will handle this scenario ?

/Tzvika

MichalPrincNXP, 2020-11-25:
Hi, I am afraid issuing erpc requests simultaneously from two tasks is not possible, unless there is any mechanism implemented on the application level that would allow requests serialization. Consider wrapping erpc calls in each task by an inter-task mutex lock that would ensure single request-response communication is performed as one operation.

tmordeko, 2020-11-25:
Hi, Thanks @MichalPrincNXP this is what I suspected :-) 
I saw that there is client arbitrator for sharing client-server communication channel isn't there a similar mechanism to handle multiple server requests, working in the granularity of a single erpc message, meaning the server will be able to process multiple messages but not necessarily associated with the same erpc request  (in OS environment off course)

Thanks in advance,
Tzvika

Hadatko, 2020-11-25:
Technicaly i think you can have several server objects but every server need its own transport (communicate trough different transport).

tmordeko, 2020-11-25:
Thanks @Hadatko , technically the same client instance can be used from multiple tasks.

Hadatko, 2020-11-25:
Yes, but you should wait until one rpc call will be completly done until call another one. Maybe you can use oneway type function where client is not waiting for answer. And get ouputs later. Depends on your use case...

tmordeko, 2020-11-26:
Thanks all for the answers

closed 2020-11-26 by tmordeko