EmbeddedRPC/erpc #161: Compiling eRPC 1.8.0 for microcontroller on Windows

69ronaldinho420, 2021-02-09:
Hello,

I am trying to compile erpc_c 1.8.0 using S32 Design Studio for a S32K144 microcontroller on Windows, but this seems to not be supported. The issues that i keep hitting are headers like termios.h, poll.h etc that are not available on Windows.

I have an older version, erpc_c 1.7.3 that compiles successfully on Windows with S32DS. The question is, has support for compilation on windows for erpc_c been dropped since? I would like the new version for the native windows serial port (i have an use case where i want to have s32k144 as server and windows as client)

Thank you

Hadatko, 2021-02-10:
Could you post compile issue related to erpc?

Anatoriann, 2021-04-06:
Hello, here are the error we currently have

[image]

MichalPrincNXP, 2021-04-23:
Hello @69ronaldinho420 , @Anatoriann , 
the eRPC primary focus is the embedded. I am afraid windows support in this library is not official and has been integrated via this PR #108 . You could contact the PR author to discuss this.

As for the case when there is eRPC server running on the embedded and the eRPC client on the PC side, you can consider using python serial transport. The example of such a configuration is provided in selected NXP MCUXpressoSDK packages. Once downloading for instance the FRDM-K64F or FRDM-K82F packages from the [https://mcuxpresso.nxp.com](https://mcuxpresso.nxp.com)  you should find the erpc_client_matrix_multiply_uart example in <MCUXpressoSDK_install_dir>/boards/<board_name>/multiprocessor_examples that demonstrates usage of eRPC between PC and board using UART transport layer.  The run_uart.py is located in <MCUXpressoSDK_install_dir>/boards/<board_name>/multiprocessor_examples/erpc_common/erpc_matrix_multiply. This could be a Python-based alternative for you, hope it helps.

Regards
Michal

closed 2022-05-08 by Hadatko