EmbeddedRPC/erpc #166: erpc_server_poll on non rpmsg transports?

joakim-tjernlund, 2021-02-19:
In https://github.com/EmbeddedRPC/erpc/wiki/eRPC-infrastructure is says:
```
The simple server contains 2 functions for message processing: run and poll. The poll function only supports the RPMsg transport layer.
```

I am speculating that the problem lies in hasMessage() and that a simple sequence counter in erpc header would
make it possible for hasMessage() to do the right thing. Have you considered adding a seq counter?

MichalPrincNXP, 2021-02-23:
I guess this is a documentation issue, NXP SDK examples show erpc_server_poll() function use with UART/SPI transport layers. @Hadatko , could you confirm, please?

Hadatko, 2021-02-23:
I think it is  "catdog" :D. Poll is working in way that it is called once. But transport layers are blocking. So poll will hang in receive function until message will arrive. 
RPMSG is using ringbuffers. So in this case you will return back immediatly if no message is in queue. 

We can think about eRPC ring buffer system...