EmbeddedRPC/erpc #167: Server connection to 2 different clients

lrodriguez14, 2021-02-23:
Hello,

I have a simple client-server connection, if is it possible to connect another client from another side using the same transport layer (TCP/IP)? If it is possible, should I use the code arbitrator?

|                     |     TCP/IP              |                     |      New connection (TCP/IP)  |                    |
| Main Client |------------------ >|     Server      | <------------------------------ | New client  |
|                     |                             |                     |                                                |                    |

MichalPrincNXP, 2021-04-23:
Hi, I think it should work in case of TCP transport, even without the arbitrator running on the server side. @Hadatko , could you confirm, please? Thanks.

Hadatko, 2021-04-26:
Hi, actually not sure if it will work client---> server <--- client. But client ---> server, server <---- client should work definitely i think. In second case you need create two identicall server objects in their threads with different ports.

drumairali, 2021-09-25:
i am trying too hard to find a single working example in C. basically i want to send data from raspberry pi to PC. using matrix multiplocation example it works but i dont have a working C example. i want to run C on client side and Server side and i want to do serial communication.... 
please make a single working example of matrix multiplication in linux using C.... 
thank you

MichalPrincNXP, 2021-09-27:
There is the test code in CPP for serial communication testing in test/common folder , see 
[unit_test_serial_client.cpp](https://github.com/EmbeddedRPC/erpc/blob/develop/test/common/unit_test_serial_client.cpp)
[unit_test_serial_server.cpp](https://github.com/EmbeddedRPC/erpc/blob/develop/test/common/unit_test_serial_server.cpp)
This is not maintained but could help you.