EmbeddedRPC/erpc #172: Signature mismatch between erpc::Transport::send and erpc::InterThreadBufferTransport::send

yoshinao-takada, 2021-03-20:
The two functions had different signatures and I could not instantiate InterThreadBufferTransport object.
Could you please correct definitions of the send() methods. I thought that the definition of InterThreadBufferTransport was
adequate but most of Transport children classes have the definitions same as Transport base class. I could not determine which should be correct, 'const' or no 'const'.

MichalPrincNXP, 2021-04-26:
Hello @yoshinao-takada , thanks for reporting. The InterThreadBufferTransport is not used much now (kind of obsolete). Anyway, it should has the send method with no-cost as the Transport class defines: virtual erpc_status_t send(MessageBuffer *message) = 0;
I will fix it.

MichalPrincNXP, 2021-05-07:
solved by https://github.com/EmbeddedRPC/erpc/commit/f99e25233370752c6c9f88390e6652a271ff617b

closed 2021-05-07 by MichalPrincNXP