EmbeddedRPC/erpc #176: Asynchronous events support

zaporozhets, 2021-05-13:
Hi All!

I noticed that [temp_alarm.erpc](https://github.com/EmbeddedRPC/erpc/blob/develop/examples/temp_alarm.erpc#L70) mentions `Asynchronous events from M4 to Linux.` But I could find erpc transport(serial, tcp, shared mem, etc.) or API that supports this feature. Does erpc support asynchronous events(from server to client) or it's something that will be added later?

Cheers,
Taras.

Hadatko, 2021-05-13:
Hi @zaporozhets , asynchronous messages (do not expect respond) are done trough usage of "oneway" keyword in front of function declaration.

Hadatko, 2021-05-13:
Sorry my first comment was about asynchronous calls (one direction).
What example is pointing to is have one client,server on m4 side and another client,server on Linux side. So Linux may asynchronously send messages to m4 and m4 to Linux. You need use client arbitrator instead of basic client implementation.

closed 2022-05-03 by Hadatko