EmbeddedRPC/erpc #182: Occasional Hangup with Serial Transport

ramlco, 2021-05-21:
Hello,

I am experiencing an occasional hangup with my application.  I am using the Windows Serial transport on the Client side and my own USB-CDC port on the Server side.  Normally, when the Sever is sending a response to a Client request, I see the following:

1. Sever sends a 4 byte header
2. Client performs serial read, gets 4 bytes
3. Server sends the message (16 bytes in this case)
4. Client performs serial read, gets 16 bytes
5. All is well, Client moves on

In the error state, I see this:
1. Server sends a 4 byte header
2. Server sends the message (16 bytes in this case)
3. Client performs serial read, gets 20 bytes
4. Client is now stuck in the 'serial_read' function because it read more than the 4 bytes is was expecting

To fix this, I would propose changing the 'nNumberOfBytesToRead' paremeter in the 'ReadFile' call to use the requested read size as a maximum rather than the Rx buffer size.  Here is a screenshot of the fix (line 226):
[image]

Thanks in advance!  This has been a great application to work with!

Hadatko, 2021-05-21:
Hi @ramlco, thank you for the propose. I am not sure if this is original file or we (community) changed it already. Maybe we should update file with source. Unfortunately the person who put that file into project is no longer active. I guess this can be related source: https://github.com/BusPirate/Bus_Pirate/blob/b580ca9dcda14b7c64b6be2ed1b4b376366b1738/scripts/powertools/SPISniffer/linux-version/serial.c

Hadatko, 2021-05-21:
@ramlco feel free to create PR with your fix and i will create new issue to update that file

aelray, 2021-06-25:
Had the same issue, this would be fixed with  #190