EmbeddedRPC/erpc #185: Make erpc in MINGW in Windows 10

kolodych, 2021-05-25:
Hello.
I have problems with compilling in MINGW. When I try to "make install" next error appears:  
```
erpc/erpc_c/port/erpc_serial.cpp:210:19: error: unused variable 'bread' [-Werror=unused-variable]
     unsigned long bread = 0;erpc/erpc_c/port/erpc_serial.cpp:309:70: error: 'strnlen_s' was not declared in this scope
         memcpy(full_path, port, strnlen_s(port, sizeof(full_path) - 1));
                                                                      ^
At global scope:
cc1plus.exe: error: unrecognized command line option '-Wno-deprecated-register' [-Werror]
cc1plus.exe: all warnings being treated as errors
```
What I am doing wrong?)

Hadatko, 2021-05-25:
Hi @kolodych so far as i know mingw was supported only for small period of time and it is not for very long time and it was never tested on Windows10

kolodych, 2021-05-25:
Thanks for the answer.
What can I use to compile in windows 10 ?

Hadatko, 2021-05-25:
ERPC lib is not available under Windows. You need create your project with transport you want to use on windows. Currrently maybe serial is only supported. Maybe tcp. I am not sure.

Hadatko, 2023-09-27:
https://github.com/EmbeddedRPC/erpc/pull/344
https://github.com/EmbeddedRPC/erpc/pull/365

closed 2023-09-27 by Hadatko