EmbeddedRPC/erpc #186: python中erpc_test文件不存在

Peanut0709, 2021-06-02:
python中 import erpc_test不存在该文件

Hadatko, 2021-06-02:
Is this ticket related to missing erpc_test module here? https://github.com/EmbeddedRPC/erpc/blob/1bb6e22a7a1f7e21eb300dd547ddfd6c15ae8d9f/erpc_python/test.py#L16

MichalPrincNXP, 2021-06-24:
Hi @Peanut0709 , could you be more specific, please? Thanks.

Peanut0709, 2021-07-26:
> 嗨@Peanut0709，你能更具体点吗？谢谢。

test.py这个文件import erpc_test   但是 整个工程里面没有erpc_test这个文件所以会报错

Peanut0709, 2021-07-26:
> 嗨@Peanut0709，你能更具体点吗？谢谢。

运行后显示  No module named 'erpc_test'

Peanut0709, 2021-07-26:
> 此票是否与此处缺少erpc_test模块有关？
> 
> https://github.com/EmbeddedRPC/erpc/blob/1bb6e22a7a1f7e21eb300dd547ddfd6c15ae8d9f/erpc_python/test.py#L16

drumairali, 2021-09-09:
after installing setup.py when you execute test.py 
it says "no module name erpc_test" what is its solution ?

Hadatko, 2021-09-09:
It looks like files like test.py, run_test.sh are obsolite or not all files were published. If they are obsolite also README in same folder should be updated

Hadatko, 2021-09-09:
Instead you can try this example which i tried and it is working:
https://github.com/EmbeddedRPC/erpc/tree/develop/examples/matrix_multiply_tcp_python

MichalPrincNXP, 2021-09-13:
> 
> 
> It looks like files like test.py, run_test.sh are obsolite or not all files were published. If they are obsolite also README in same folder should be updated

Thank you, @Hadatko , yes these files are obsolete and I will delete them.
@Peanut0709 , @drumairali , please use the python example @Hadatko  is referencing.

drumairali, 2021-09-22:
Can I get any example in which in which from python remote procedure call
is made and on C side it executes and returns results back on python side?.

MichalPrincNXP, 2021-09-24:
Hello @drumairali I am afraid we do not have such an example (combining Python on one side and the C on the other side) in the eRPC repo. 
BUT, there are such examples in NXP MCUXpressoSDK packages for selected boards. Once downloading for instance the FRDM-K64F or FRDM-K82F packages from the https://mcuxpresso.nxp.com you should find the erpc_client_matrix_multiply_uart example in <MCUXpressoSDK_install_dir>/boards/<board_name>/multiprocessor_examples that demonstrates usage of eRPC between PC (Python client impl.) and board (C server impl.) using UART transport layer. The run_uart.py is located in <MCUXpressoSDK_install_dir>/boards/<board_name>/multiprocessor_examples/erpc_common/erpc_matrix_multiply. See the associated readme for this example. Hope it helps.

drumairali, 2021-09-24:
like we do matrix multiplication in Python. Is there any working code with which we can do same in C. may be if i will have that C matrix multiplication code then on server side i can run C and on Client side i can run python and i can send the data from Py to C?

drumairali, 2021-09-29:
thank you so much,,, i wanted to use Raspberry Pi (python) and PC (C/C++) and i wanted to do serial communication using erpc. Right now i am able to do eRPC between Raspberry Pi and PC via rs232 cable but both sides are in python. I want to run a test program in which on PC i just want to do same matrix multiplication example in C.

so is there any way in which i can run C/C++ matrix multiplication program on PC using eRPC exactly like i can do in python (i need this python example(https://github.com/EmbeddedRPC/erpc/tree/develop/examples/matrix_multiply_tcp_python) in C/C++ ). This example has option for both Serial and TCP transport bdw. Can i do it on ubuntu in C/C++ because i have been on it for last 15+ days and no outcome..

MichalPrincNXP, 2021-09-29:
Hello @drumairali , your use case is valid but not demonstrated in any example. You have to build it by your own. When intended to use the serial transport on PC, look at and add into your project the [erpc_serial_transport.cpp](https://github.com/EmbeddedRPC/erpc/blob/develop/erpc_c/transports/erpc_serial_transport.cpp), there is also the C setup function once you would use C code, but this seems to be not used/modified a long time ([erpc_setup_serial.cpp](https://github.com/EmbeddedRPC/erpc/blob/develop/erpc_c/setup/erpc_setup_serial.cpp)).
I would also recommend to have a look at [unit_test_serial_client.cpp](https://github.com/EmbeddedRPC/erpc/blob/develop/test/common/unit_test_serial_client.cpp) / [unit_test_serial_server.cpp](https://github.com/EmbeddedRPC/erpc/blob/develop/test/common/unit_test_serial_server.cpp) to see how to setup erpc with serial transport in C++ code. Hope it helps.

drumairali, 2021-09-29:
thank you so much, i appreciate your quick response... i try it..

Hadatko, 2023-09-27:
It looks like this thread can be closed

closed 2023-09-27 by Hadatko