EmbeddedRPC/erpc #188: Multiple Clients Solution

yahya010, 2021-06-18:
Hi,

I wanted to know how a multiple client solution for erpc would be implemented and what changes would I need to do. Also, if its not possible, how can I do it using TCP only?

Thanks,

Hadatko, 2021-06-18:
Hi @yahya010 , please take a look to https://github.com/EmbeddedRPC/erpc/issues/174

vonku, 2023-03-14:
Hi @Hadatko, I‘m so looking forward to the feature, will it be released to develop branch this month?

Hadatko, 2023-03-14:
Hi @vonku , Although supporting multiple client is almost ready, i need to appologize that i didn't focus on it last months. I distracted myself with CI which is also very important to properly test every our change we do. Now CI support build and test erpcgen under Mac, Linux, Windows. This can help us to do less mistakes and better releases.
Right now i don't see more important task than this (multiple client support.)

vonku, 2023-03-15:
> Hi @vonku , Although supporting multiple client is almost ready, i need to appologize that i didn't focus on it last months. I distracted myself with CI which is also very important to properly test [...]

Hi @Hadatko , I fetched your pr and tried it out lastday. I found it different that the IDL annotation **@types_header(string)** did not work, with which I wanted to declare all the typedefs and structs to a specific header file to avoid redefinitions. Now they were generated to  xxx_common_erpc.h and xxx_common_standard.h. Is this change on purpose?

vonku, 2023-03-15:
> Hi @Hadatko , I fetched your pr and tried it out lastday. I found it different that the IDL annotation **@types_header(string)** did not work, with which I wanted to declare all the typedefs and [...]

As shown below, I hope that the structs defined in _bt_common.erpc_ would be generated to bt_erpc_common.h, but no bt_erpc_common.h file was generated. 
```
@output_dir("bt_source")
@types_header("bt_erpc_common.h")
program oppo_bt_erpc

import "bt_common.erpc"
```

Hadatko, 2023-03-15:
Hi @vonku, i think i removed it intentionally as the type definitions are now generated into their own files by default. So i can add two bullets to my work:
-> Type definition should be generated only into one file. They may not depend on interface (like in IDL they are not depending on interface). 
-> Keep option for renaming the file (Do you think it is necessary?).

vonku, 2023-03-15:
Another issue, If I define the ERPC_ALLOCATION_POLICY  as below, the shim code would compile failed.
`#define ERPC_ALLOCATION_POLICY (ERPC_ALLOCATION_POLICY_STATIC)`

Because the erpc_port.h file would not be included in xxx_client.cpp file but it still uses `erpc_malloc()` and `erpc_free()`
```
#if ERPC_ALLOCATION_POLICY == ERPC_ALLOCATION_POLICY_DYNAMIC
#include "erpc_port.h"
#endif
```

amgross, 2023-03-15:
Sounds like the malloc issue @diagano had at #349

vonku, 2023-04-08:
Hi @Hadatko, is there any update plan for this feature?

Hadatko, 2023-04-08:
Hi @vonku , same as last time. In eRPC is this my main issue right now (finish this task). But eRPC is not my only project. So i couldn't work on it last months.

closed 2023-10-17 by MichalPrincNXP