EmbeddedRPC/erpc #189: Feature Request:

aelray, 2021-06-18:


MichalPrincNXP, 2021-06-25:
Hi @aelray , there is no simple way how to do it using any settings or configuration. You could try to modify the erpc_framed_transport.cpp/.h code directly, disabling m_crc calculation in the send method and also the crc checking in the receive method. This is quite easy adjustment, I think. Hope it helps.
Regards
Michal

kikass13, 2021-06-26:
shouldn't anyone provide a configuration class for erpc, so that these things can be configured properly at runtime/buildtime instead of suggesting that people just fork their own erpc version to just change a minor thing?

MichalPrincNXP, 2021-06-28:
@aelray , attached is the updated code with the new ERPC_CRC config macro (needs to be set in your erpc_config.h file) to try on your side. Note, this has not been fully tested. Please let me know if this is sufficient for you? Thank you
Michal
[erpc_c.zip](https://github.com/EmbeddedRPC/erpc/files/6725512/erpc_c.zip)

aelray, 2021-07-01:
> @aelray , attached is the updated code with the new ERPC_CRC config macro (needs to be set in your erpc_config.h file) to try on your side. Note, this has not been fully tested. Please let me know [...]

Thank you! I will try when I have time

Hadatko, 2023-08-28:
Hello, i will do version without CRC. I would like to do it without macro, maybe just with inheritance.