EmbeddedRPC/erpc #192: erpc generic device transport layer

kikass13, 2021-06-20:
Hello,

I have integrated erpc into our embedded platform (modm + arm) and successfully ported the matrix_multiplication example.

I have looked into the transport layers and successfully iplemented a fifo transport (local fifos) for my server and client application (on the same device).

There are discussions about having erpc abstract multiple different buses in our system, but some things seem a little bit odd to me. 

As of now, everyone just implements a random protocol driver into erpc transports for bus communication. This, in my opinion, is a major design flaw and I am confident in saying, that erpc transports should not inherit the burden of device drivers (CAN, ETHERNET, UART, etc.) because these are hardware dependent and therefore subject to migration and porting.

Maybe I did not see it, but i would suggest putting a generic device driver as a soft wrapper inside the transport layers, which has a well defined interface and can be injected from user code. I have implemented something like this (successfully) and maybe someone here is interested in a discussion about such a feature (if it does'nt exist already somewhere and Im just blind :) ).

A very brief suggestion of what Im talking about here:

```cpp

/// define my device wrapper class, which just serves as an entry point from inside erpc and does the complex message handling and driver things from inside my user code

class MyDeviceWrapper : public Device
{
public:
    FifoDeviceWrapper() : buffers_()
    {}
     
     bool hasReceived() final 
       {
             /// implement whatever it takes to "acknowledge receiving data" via whatever protocol you want 
            /// utilizing whatever driver we want without putting garbage into erpc
	}

    erpc_status_t connect()
    {
            /// implement whatever it takes to "connect" via whatever protocol you want 
            /// utilizing whatever driver we want without putting garbage into erpc
    }

    erpc_status_t write(const uint8_t* data, const uint32_t size) final
    {
             /// implement whatever it takes to "write" via whatever protocol you want 
            /// utilizing whatever driver we want without putting garbage into erpc       
    }

	erpc_status_t read(uint8_t* data, const uint32_t size) final
	{
            /// implement whatever it takes to "read" via whatever protocol you want 
            /// utilizing whatever driver we want without putting garbage into erpc
        }
};

int main()
{
    /// erpc pre stuff
    /// ...
    MyDeviceWrapper device();
    erpc_transport_t c_transport = erpc_transport_modm_device_init(&device);
} 

```

The device will be called from inside the transport layer of erpc and I can now implement drivers on top instead of polluting the relatively simple erpc transport with driver based stuff.

Am I missing something? would something like that help ?

kikass13, 2021-06-26:
I have drawn a picture, to better show what I have implemented successfully using the CAN protocol
 * 1 stm32 erpc client calling matrix_multiplication rpc via CAN
 * 1 stm32 erpc server serving matrix_multiplication rpc via CAN

[image]

This allows me to 

* A: not keep drivers for specific devices inside erpc (which is a design flaw in my opinion) 
* B: have the erpc specific transport implementation in my application code (better maintainability) 
* C: have the ability to interact with the erpc communication and/or hijack / debug whatever is going on there

MichalPrincNXP, 2021-07-01:
Hello @kikass13 
thank you for the proposal. The erpc already has abstracted transport interface and it is designed to allow easy transport change by just replacing two transport source files (cpp/h) in the user project. To be honest I do not see the benefit of the proposed approach - having multiple different device specific transport implementations in the app. or in the erpc code seems to me as the same. Providing selected transport implementations in the erpc repo can serve also as the example how to implement your own device specific transports when needed. 
Regards
Michal

kikass13, 2021-08-09:
@MichalPrincNXP 

well erpc (as a repository) does not build for target platforms. You need to manually alter/copy (or cmake magical unicorn) whatever works for your project. The result is `erpc` which 
 * never looks identical (for different users) 
 * is not forkable properly, because everyone just massacres the original repository
     *  look at all those wonderful forks, aren't they pretty and they will never be merged  /sarcasm 
 * is full of hardware/platform specific stuff which does not belong there

I accept your answer, but that does not fix the fact, that this library needs a proper abstraction layer which shields it from outside use cases (providing a proper transport layer API to the OUTSIDE of erpc, not inside of it). 

A proper use case is to derive from a transport layer class and implement on top of it (in user space), not fork the repository and try to hack whatever you want into it because of some really important special factoriy functions.

My Device implementation example sits on top of your transport layer, but it should just replace it completely.

Hadatko, 2021-08-09:
@kikass13 
Sorry i still don't see difference between what you suggest and what we propose. Current transport abstraction is proposing same advantage and it can be easily changed in code. So maybe we need example what we cannot do and how you would solve that to understand your use case...

kikass13, 2021-08-09:
@Hadatko 
Thanks for the answer.
Hmm then maybe I have to retry my example again and see if my concerns are invalid :)

I am mostly concerned with the fact that erpc consists of these "setup_whatever_transport" functions, as these (in combination with the appropriate transport implmenetation) seem to be out of place

Hadatko, 2021-08-09:
@kikass13 Primary we are supporting C language using C++ features. That is why setup functions exists. As they may  be called from C.

kikass13, 2021-08-09:
@Hadatko yes i will take another look into it all and be back if something comes up :), thanks for the help

closed 2021-08-09 by kikass13

Hadatko, 2021-08-09:
You are welcome ;)