EmbeddedRPC/erpc #198: Mu receive Disable Interrupt missing when Rtos is not used.

Hadatko, 2021-07-22:
Current:
```C
#if !ERPC_THREADS_IS(NONE)
            MU_DisableInterrupts(m_muBase, (1U << (MU_CR_RIEn_SHIFT + MU_RR_COUNT - MU_REG_COUNT)));
            m_rxSemaphore.putFromISR();
#endif
```

Should be:
```C
MU_DisableInterrupts(m_muBase, (1U << (MU_CR_RIEn_SHIFT + MU_RR_COUNT - MU_REG_COUNT)));
#if !ERPC_THREADS_IS(NONE)
            m_rxSemaphore.putFromISR();
#endif
```

MichalPrincNXP, 2021-07-26:
Yes, will fix it, thank you.

MichalPrincNXP, 2021-11-16:
Hi Dusan, you have moved the MU_DisableInterrupts() function call before the conditional section within the #158. This seems to be reasonable but then the baremetal MU-based erpc example stopped working, so I have reverted that change. So, now all works fine on my side. Do you have any project that fails with the current MU transport implementation? To be honest I do not have the explanation why disabling the MU interrupts in baremetal causes the example malfunction (the erpc matrix multiply request is processed just once, no other request captured on the server side).

Hadatko, 2021-11-16:
Do you know why baremetal version need enable RX interupts ? Maybe this may help us. But it looks like for some reason we can't disable rx interrupts. Maybe we will miss it (it is only for short time?). Then we should also get rid of enabling interrupts for baremetal. And keep it enabled forever as it is anyway. But i would like to know exact reason. 
[image]

MichalPrincNXP, 2021-11-16:
It seems the bm is not the problem but the used poll mechanism in the baremetal example. Once erpc_server_run() is used instead of the erpc_server_poll() then all works fine with the MU_DisableInterrupts() function being called before the conditional section. So, there must be an issue in the mechanism that tracks if a new message has been received, i.e. in hasMessage() method / m_newMessage variable. I am afraid the m_newMessage is not handled correctly.

MichalPrincNXP, 2021-11-16:
Once hasMessage() in erpc_mu_transport.h implemented as "return true" (always return true) polling works fine ... may I ask you to review the m_newMessage variable handling, please?

Hadatko, 2021-11-16:
I think in current implementation we need revert Disabling interrupt position and get rid of Enabling interrupt for rx for baremetal.
The reason is this: 
[image]

This way it is nonblocking for polling. Blocking way for poling would be set hasMessage to return True and in recv function loop until interrupt occur (or return something like NoMessageReceived.).

Hadatko, 2021-11-16:
There are missing some important comments in this driver. But my change was wrong and to code has to be as it was originally. 
FREERTOS: blocking: Enabling interrupt on recv start, get data once ready, disable interrupt on receive end.
BM: nonblocking: enable interrupt, wait for interupt, once interrupt appeard disable interrupt and set hasMessage==true, recv start,enable interrupt, get data, receive end (keep interrupt enabled for same reason as first time)

MichalPrincNXP, 2021-11-19:
Thank you, Dusan, for this analysis. I will put your comment into the erpc_mu_transport.cpp and we could close this issues, ok?

Hadatko, 2021-11-19:
Yes. Sorry for this long False/Positive bug.

MichalPrincNXP, 2021-11-19:
No, no, that helped to understand the implementation better ... thank you.

closed 2021-11-19 by MichalPrincNXP