EmbeddedRPC/erpc #209: error handling on erpc server exit

bastivi, 2021-10-13:
Hi,

I'm using following setup:
`*---------------*            *-----------------*
|  erpc Client  |     ---->  |  erpc Server    |
*---------------*            *-----------------*
 - Xilinx UltraScale+         - NXP LPC54605
 - petalinux                  - FreeRTOS
 - SPI Master                 - SPI Slave`

I've problems to do proper error handling if the erpc server fails with e.g. CRC error. Then I have to restart the erpc server again. But the erpc client stuck into endless loop within erpc lib (erpc_spi_master_transport.cpp - SpidevMasterTransport_WaitForSlaveReadyGpio()) waiting for the markers transferred via SPI. The erpc server do not release this loop (do not send markers) on restart so it is hard to get them out there.

What’s the correct way doing this or is there a erpc server to erpc client function missing to tell the erpc client to abort this request?

It's also already posted on NXP:
https://community.nxp.com/t5/LPC-Microcontrollers/eRPC-error-handling-for-SPI-transfer/m-p/1354498/highlight/false#M46747

Thank you a lot and best regards
Sebastian

MichalPrincNXP, 2021-10-15:
Hello Sebastian,
thanks for this report. The comment has been provided via the NXP community already but I agree the solution is not ideal in this case. There is no mechanism inside the erpc that would cover that transport error situation. The rootcause is in the SPI transport layer implementation that uses blocking SPI driver API without the timeout possibility. We should think of this transport rewrite with non-blocking API and the timeout possibility. I would recommend now to use any other mechanism for the error state signalling like you do with the GPIO.

bastivi, 2021-10-18:
Hi,
thank you for your answer. I'll go on with the GPIO signalling, anyway, if there is the possibility to use GPIO for this task this is the more performant solution.