EmbeddedRPC/erpc #214: Multiple servers running on one process

Qeenyra, 2021-11-03:
Hi,
When creating multiple servers, it looks like g_server in erpc_server_init will get overwritten with each new server init.
Is it possible to have more than one server running on one process?
Thanks!

MichalPrincNXP, 2021-11-05:
Hello @Qeenyra , I am afraid multiple servers are not supported yet.

Hadatko, 2021-11-08:
Hi, there can be multiple resource independent servers run. Just do not use default setup implementation for server. You can get inspiration from there to create as many servers as you want ;)

Hadatko, 2021-11-08:
I am thinking that we can change implementation of this function to add to each function server id. And define in erpc_config the amount of servers. This way we can provide this functionality to users.