EmbeddedRPC/erpc #217: Arbitrated server/client between a SPI with a handshake line

ftab, 2021-11-12:
Working on a two microcontroller eRPC setup between ESP32 and TM4C.

Both microcontrollers will need to call functions on demand, I'm using the TM4C as a GPIO expander because the ESP32's out of pins and doesn't have USB. So the TM4C will need to call a function to give received CAN and USB frames, ESP32 will need to call a function to send frames out the CAN or USB, read or set GPIOs, etc. TM4C is the master because that was the highest speed available between the possible configurations (10mhz, vs. something like 6mhz in the other direction)

TM4C is the SPI master, and ESP32 has a GPIO set as the "ready" pin. We wrote the TM4C equivalent of the spi master transport, and the ESP32 equivalent of the spi slave transport. Operating the SPI in half duplex, essentially (full duplex peripheral mode, but just throwing away the half that isn't being used in each underlyingSend and underlyingReceive)

We have the TM4C client and ESP32 server bare minimum example running, so it can call a one-way function on ESP32 when a button is pressed on TM4C.

It seems like client/server setup on both ends is the best bet, with the transport arbitrator, so that the functions can be called on demand at any time without having to poll.

Initially both are in an idle ready-to-receive state. There is a button on the TM4C that will execute the same one-way function as before.

When is the ready pin supposed to go high, and when does it need to go back low? How does each side decide when it's time to tx vs. time to rx? Is there a default header that we should expect when reading in a state where it doesn't need to do anything?

Do we need a second handshake line that indicates it's ready to tx something so the other side needs to rx?

Thanks for any guidance you can give.

MichalPrincNXP, 2021-11-15:
Hello @ftab , I would recommend to have a look at erpc_client_matrix_multiply_spi / erpc_server_matrix_multiply_spi example projects that are provided in NXP [MCUXpressoSDK](https://mcuxpresso.nxp.com/en/welcome) packages ... you can pick the frdmk64f board, enable the multicore middleware and once the package is ready go to boards\frdmk64f\multiprocessor_examples\ folder. You should be able to get answers to your question re. the GPIO use for handshaking. Note, this example does not use the arbitrator, I guess that the SPI transport will unfortunatelly not work correctly with the arbitrator because of missing receive messages queue in the SPI transport (gab identified in other transport).

ftab, 2021-11-15:
> I guess that the SPI transport will unfortunatelly not work correctly with the arbitrator because of missing receive messages queue in the SPI transport (gab identified in other transport).

is there a physical transport available within a few pins that would work in this manner?

MichalPrincNXP, 2021-11-15:
I am afraid the transport arbitrator works fine with rpmsg_lite and TCP transports only as these transports provide incoming messages buffering.

ftab, 2021-11-17:
Thanks! I think we'll try adding a UART to the mix for the other direction, since we have a couple of pins left.

closed 2021-11-17 by ftab