EmbeddedRPC/erpc #230: Error with erpcgen in generating list<binary>

arnoldgb-arm, 2022-01-12:
struct test_t{
  list<binary> var
}

When you generate cpp files for above erpc file, the erpcgen doesn't generate 'read_binary_t_struct' for *server.cpp and 'write_binary_t_struct' for *client.cpp. This causes the compilation to fail. Need help with this.

Hadatko, 2022-01-12:
Hello, thank you for your bug report. Feel free to create PR. Otherwise i can look on it once i will have time for that (few weeks)

Hadatko, 2022-02-05:
> struct test_t{ list var }
> 
> When you generate cpp files for above erpc file, the erpcgen doesn't generate 'read_binary_t_struct' for *server.cpp and 'write_binary_t_struct' for *client.cpp. This causes the compilation to fail. Need help with this.

Can you show me an example how you are using this type in your IDL file? If it is not used than no serialization code may be generated.

correct IDL declaration
'''C
interface A{
      oneway fun(binary T)
}
'''

correct IDL declaration 2
'''C
struct test_t{
binary var
}

interface A{
      oneway fun(test_t T)
}

'''
correct IDL declaration 3
'''C
struct test_t{
list<uint8_t> var
}

interface A{
      oneway fun(test_t T)
}
'''

Generated files looks good to me.

arnoldgb-arm, 2022-02-05:
```
struct test_t{
list<binary> in
}

interface handler {
  fun(test_t V) -> int32
}
```
Doesn't work for the above.

Hadatko, 2022-02-05:
@arnoldgb-arm Could you verify if it is still an issue? It looks ok for me with current eRPC version. See zip bellow: 
[acrios-erpc.zip](https://github.com/EmbeddedRPC/erpc/files/8008258/acrios-erpc.zip)

arnoldgb-arm, 2022-02-05:
```
struct test_t{
list<binary> a
}

interface handler {
  fun(test_t V) -> int32
  get() -> test_t
}
```
This was the one causing error. Sorry for the confusion

Hadatko, 2022-02-05:
@arnoldgb-arm Now i see and confirming the bug.

Hadatko, 2022-02-05:
@arnoldgb-arm as temporary solution you can use:

'''C
struct test_t{
list\<uint8\> a
}

interface handler {
  fun(test_t V) -> int32
  get() ->test_t
}
'''

Hadatko, 2022-02-05:
Actually this solution looks not working too. So the issue is more generic than i thought. I will try to fix this asap.

Hadatko, 2022-02-05:
I found a bug and gods are merciful as it looks like i found also solution which looked be complicated, but it is simple.

Hadatko, 2022-02-05:
PR ready for merge. Thank you for noticing us with this issue. You may build erpcgen with this bugfix on your own or it will be awailable (i hope) at next release.

arnoldgb-arm, 2022-02-05:
> PR ready for merge. Thank you for noticing us with this issue. You may build erpcgen with this bugfix on your own or it will be awailable (i hope) at next release.

Thank you!

closed 2022-02-07 by MichalPrincNXP