EmbeddedRPC/erpc #241: Calling memcpy with invalid pointer

amgross, 2022-02-06:
Hi,
In c autogenerated code (with BasicCodec), in case that read binary fails (and change the m_status to failure), the auto generated calls memcpy with length 0 and valid dst pointer, but with unknown src pointer (no one defined $info.dataTemp). According c spec this may lead to undefined behavior.
https://github.com/EmbeddedRPC/erpc/blob/21a396ed9ad3cfd5597ac75bb905d68f0d4eb7ad/erpcgen/src/templates/c_coders.template#L35-L56
Optional solutions will to have version of erpc_malloc that gets codec and returns NULL if the status is not OK (that skips the memcpy), or not do the malloc + memcpy at all if status is not success

Hadatko, 2022-02-06:
Hi @amgross , do you think this will fix it for you? https://github.com/EmbeddedRPC/erpc/pull/72

Hadatko, 2022-02-06:
Likely not. I will look on it if it is an issue. Thank you for your feedback.

Hadatko, 2022-02-26:
Hi @amgross , i agree with you. I will propose solution soon.

Hadatko, 2022-02-26:
Hi @amgross , just to be sure i did this change #253 . But based on talk here: https://stackoverflow.com/questions/3751797/can-i-call-memcpy-and-memmove-with-number-of-bytes-set-to-zero
it looks like length 0 should be ok. 
Do you agree with changes or you would like to improve that?

amgross, 2022-02-27:
Hi @Hadatko , I disagree that 'NULL should be valid ptr value'.
You can see first answer at https://stackoverflow.com/questions/5243012/is-it-guaranteed-to-be-safe-to-perform-memcpy0-0-0 that claims that 'results in undefined behavior, because null pointers are considered "invalid values."'.
Same you can see in one of the comments in the question you referred,  https://stackoverflow.com/questions/3751797/can-i-call-memcpy-and-memmove-with-number-of-bytes-set-to-zero#comment34300578_3751937, claiming that 'You're correct that you can't pass null pointers'

amgross, 2022-02-27:
However you can probably put there the m_cursor.get(). It will return valid pointer (that won't be used)

Hadatko, 2022-02-27:
I did it based on: Answers. In general, the only valid pointer values are NULL, and pointers to the beginning, inside, and right after, an existing object. Also note that NULL does not necessarily have to be represented as a bit-pattern with zero:s. On some architectures, -1 is a legal pointer value.

But if function is telling that it is not handled then i will rather id it lenght condition.

Hadatko, 2022-02-27:
@amgross What about to use memmove: https://www.tutorialspoint.com/c_standard_library/c_function_memmove.htm

closed 2022-03-10 by MichalPrincNXP