EmbeddedRPC/erpc #263: NULL pointer dereference

smithBraun, 2022-03-29:
It looks (not tested) like we may have null pointer reference in case that client failed to allocate codec:

1. if createBufferAndCodec failure, returns NULL:
https://github.com/EmbeddedRPC/erpc/blob/9974b80ced6ea789dcc317e4ad2ffe17a10f9bfc/erpc_c/infra/erpc_client_manager.cpp#L190-L211

2. createRequest returns RequestContext with NULL codec:
https://github.com/EmbeddedRPC/erpc/blob/9974b80ced6ea789dcc317e4ad2ffe17a10f9bfc/erpc_c/infra/erpc_client_manager.cpp#L32-L38
3. In the auto generated code, err is set and releaseRequest is called:
```
    // Get a new request.
    RequestContext request = g_client->createRequest(false);

    // Encode the request.
    Codec * codec = request.getCodec();

    if (codec == NULL)
    {
        err = kErpcStatus_MemoryError;
    }
    else
    {
        ...
    }
    // Dispose of the request.
    g_client->releaseRequest(request);
```
4. The null codec dereferenced
https://github.com/EmbeddedRPC/erpc/blob/9974b80ced6ea789dcc317e4ad2ffe17a10f9bfc/erpc_c/infra/erpc_client_manager.cpp#L213-L215

Hadatko, 2022-03-29:
Hi @smithBraun , i think you are correct. Thank you for pointing us to this issue.

Hadatko, 2022-03-29:
@smithBraun Could you review PR?

closed 2022-03-31 by MichalPrincNXP