EmbeddedRPC/erpc #268: Python eRPC nested call

Davy4fg, 2022-05-07:
Dear eRPC team,

Thanks for the great job. 

I am trying python nested eRPC call, but it seems this funtion is not implemented.
Is there a plan to develop such feature in python? 
If not, do you have any suggestion on porting this feature from C++ to python?

Thanks

Hadatko, 2022-05-08:
Hi @Davy4fg , currently this is not on my list. In C i did it a bit complicated to save memory, code, ... In python i think it can be done easier, but i would need think what is the best way. Feel free to suggest something.

Davy4fg, 2022-05-09:
Hi @Hadatko, thanks for the feedback.
After add a m_server in Class ClientManager and let m_server to run in the end of function perform_request, the nested call is working. My problem now is how to seperate the nested call and non-nested call functions, then to decide whether m_server need to run in the end of function perform_request. Do you have any suggestions?

Hadatko, 2022-05-09:
Well i am thinking if you can do it by analyzing callstack.

Davy4fg, 2022-05-09:
you mean user maintain a list containing the nested calls, and analyze callstack if match to any item of this list?

Hadatko, 2022-05-09:
I was thinking something like this. If we can find that the eRPC client call is called inside Server then it is nested call....

```python
import traceback
from inspect import getframeinfo, stack

def b():
    print(traceback.extract_stack()[1].name == "a")

    print(getframeinfo(stack()[1][0]).function=="a")

def a():
    b()

a()
```

Hadatko, 2022-05-09:
Could you share your changes? I am happy that you have working solution.