EmbeddedRPC/erpc #272: [QUESTION] increase packet size tcp

flo-th, 2022-05-11:
Hi all,

Using a python client with a C server, with tcp transport, how could i increase the tcp packet size which seems limited to 256 bytes.

Thanks in advance.

flo-th, 2022-05-11:
Hi @Hadatko, 
I have already tried to increase the value of ERPC_DEFAULT_BUFFER_SIZE and check it new value, but it never receive the data from the python client.

Hadatko, 2022-05-11:
Currently i don't see the reason. Which message buffer are you using on C? We are preparing tcp python tests which will also communicate with C tcp test. So it will be easier to test, but not sure how soon this will be available

Hadatko, 2022-05-11:
Also are you sure you are using correct config file? Sometimes people copy template modify it and they will add include path to the template origin.

flo-th, 2022-05-12:
I am using a binary message buffer on C.
Where could i investigate to see the drop of my packet before the calling of rpc method ?

closed 2022-05-12 by flo-th