EmbeddedRPC/erpc #273: eRPC using in classic autosar

Davy4fg, 2022-05-13:
Hi eRPC developers,

Is there a possibility to run eRPC in M core with classic autosar which use C programming? 

If yes, any suggestions on porting?

Thanks

Hadatko, 2022-05-18:
Hi @Davy4fg , i am not sure what classic autosar means. This application can be run with usage C language but you need also c++ compiler.

Davy4fg, 2022-05-19:
Hi Hadatko,

Thanks for the feedback.

Classis autosar is an automotive OS used in M core. It only support static allocation RAM. Project is written in C language. Our compiler is greenhills ccarm for C and cxarm for c++.
I compile the eRPC cpp file by c++ compiler and create obj files. During linking by C compiler with other C objects, it will prompt unresolved error for "new" keyword in C++ which are used in the wrapper functions.

Do you have any suggestions?

Hadatko, 2022-05-19:
It would be helpful if you share error you got.

Davy4fg, 2022-05-19:
To make things easier. I tried a simple c++ file with a class, a c++ file for c-wrapper, and a main.c. The linker result is similar.

The error is as follows,
LD         TestSuit.elf
[elxr] (error #412) unresolved symbols: 2
 __nw__FUi      from cpp_wrapper.o
 __dl__FPv      from cpp_wrapper.o

[elxr] (error) errors during processing
make: *** [..\..\MakeSupport/Global.Makefile.target.4.mk:837: TestSuit.elf] Error 1

Maybe overload C++ new and delete methods is needed? let me make a try

Davy4fg, 2022-05-19:
Thanks, Hadatko.

There's no malloc function available in classic autosar. And I overload the new funtion to locate memory in stack, build and link is ok. But mcu stuck in loading time. I don't have debugger currently, so I can't figure out the reason. 

Anyway, it is not proper to use C++ program in classic autosar. We have not develop any project in C++ with M core in automotive industry. So is it possible to provide a pure C eRPC version some time in the future?

Hadatko, 2022-05-19:
I don't think so. It would mean rewrite whole eRPC framework. For static memory allocation we are using [ManualConstructor](https://github.com/EmbeddedRPC/erpc/blob/develop/erpc_c/infra/erpc_manually_constructed.h). But shim code may generate code which is using dynamic allocation.

Davy4fg, 2022-05-28:
Build works after changing to static memory location and dropping C++ keyword. 
Close.

closed 2022-05-28 by Davy4fg