EmbeddedRPC/erpc #274: Un-needed check for null pointer before free

amgross, 2022-05-16:
At least twice I see checking of null pointer before free it:
https://github.com/EmbeddedRPC/erpc/blob/340dd882b1cd1b6dac6f10c6dec5ce92ac8d46be/erpcgen/src/templates/c_common_functions.template#L380-L383
https://github.com/EmbeddedRPC/erpc/blob/340dd882b1cd1b6dac6f10c6dec5ce92ac8d46be/erpcgen/src/templates/c_common_functions.template#L403-L406
Can't we assume that erpc_free know to handle properly null pointer like the c standard library free function?

It is not bug, just extra code and run-time.

Hadatko, 2022-05-16:
Hi @amgross , you are correct i think we can get rid of these checks and add comment to erpc_free function that NULL pointer is valid and should be handled extra inside. Which in case of calling free is not necessary...

May i let you create PR?

amgross, 2022-05-16:
OK :)