EmbeddedRPC/erpc #278: Is it possible to add defines into IDL

tmordeko, 2022-05-31:
Hi,

Is #define supported by IDL ? we are currently substituting defines with const uint32 BLA_BLA_DEF = 100; if it is not supported then it will be very helpful

Thanks in advance,
Tzvika

Hadatko, 2022-05-31:
Hi @tmordeko currently defines are not supported. Can you explain why to use defines instead of variables? Using variables is helpful for compilers to identify possible issues. 
What you can do is define variables as external and include file where you have defines for values.

tmordeko, 2022-05-31:
Hi @Hadatko ,

To avoid the extra space
Can you please elaborate on your proposal "What you can do is define variables as external and include file where you have defines for values."

Regards

Hadatko, 2022-05-31:
mydefines.h:
#define MY_FIRST_MACRO 10

idl:
@include("mydefines.h")
program ....

@external 
const uint8_t MY_FIRST_MACRO=10

tmordeko, 2022-05-31:
Thanks ! 
Please consider to add defines into IDL in the future

tmordeko, 2022-06-01:
Hi @Hadatko 

Not sure your approach is applicable to my use case, for example we use these defines (previously defined as const vars) to indicate param's max length in function's IDL declaration, for example MY_MAX_SIZE below:

@include("my_include.h")
MY_FUNC(in uint32 param1, in uint32 data_length , in binary data @length(data_length) @max_length(**MY_MAX_SIZE**))

In my_include.h

#define MY_MAX_SIZE 100

Now after moving to .h @include annotation I get the following error msg: 
"error: line XX: The parameter or member named by a max_length annotation must exist."

In previous implementation it worked:

const uint32 MY_MAX_SIZE = 100;

MY_FUNC(in uint32 param1, in uint32 data_length , in binary data @length(data_length) @max_length(MY_MAX_SIZE))

Please advise

/Tzvika