EmbeddedRPC/erpc #279: Confusion over the use of stackSize and stackPtr in Thread class constructor

berendo, 2022-06-02:
The `Thread` class [constructor](https://github.com/EmbeddedRPC/erpc/blob/932a8aa2ef603ca52a4c37d62bb3f59a60e17c05/erpc_c/port/erpc_threading.h#L101-L102) based on `thread_entry_t entry` takes `stackSize` and `stackPtr` arguments which default to `0` if omitted. The `stackSize` is then used in the `Thread::start()` methods of the [Windows](https://github.com/EmbeddedRPC/erpc/blob/932a8aa2ef603ca52a4c37d62bb3f59a60e17c05/erpc_c/port/erpc_threading_win32.cpp#L77), [FreeRTOS](https://github.com/EmbeddedRPC/erpc/blob/932a8aa2ef603ca52a4c37d62bb3f59a60e17c05/erpc_c/port/erpc_threading_freertos.cpp#L78) and [Zephyr](https://github.com/EmbeddedRPC/erpc/blob/932a8aa2ef603ca52a4c37d62bb3f59a60e17c05/erpc_c/port/erpc_threading_zephyr.cpp#L58) ports (curiously, the [Pthreads](https://github.com/EmbeddedRPC/erpc/blob/932a8aa2ef603ca52a4c37d62bb3f59a60e17c05/erpc_c/port/erpc_threading_pthreads.cpp#L76-L78) port ignores this argument entirely and just creates the thread with defaults). There also appear to be some differences in which ports create the thread in the `Thread::start()` method vs the `Thread::init()` method, but that's a separate topic.

Now, looking the TCP transport code which created a thread to listen for connections using the above-mentioned `Thread` constructor [here](https://github.com/EmbeddedRPC/erpc/blob/develop/erpc_c/transports/erpc_tcp_transport.cpp#L58), there is absolutely no place where it could possibly be setting the `m_stackSize` value to anything other than the `0` that the constructor uses by default. That in turn means that the TCP transport's listening thread would be attempted to be created with a `0`-sized stack (which maybe works in the Pthreads case since it is not used).

So, the questions are:

1. How is the TCP transport constructor supposed to influence the stack size of its listening thread (even if the stack is static, a valid non-zero size needs to be used)?
2. Is the TCP transport working just by luck and/or _only_ on Pthreads (I realize this transport is marked "mostly for testing," but I'm failing to understand how it could be working even for the most basic testing cases on anything other than Pthreads)?

berendo, 2022-07-16:
> Hi, i will look on your question asap ;)

One more nudge @Hadatko?

Hadatko, 2022-08-01:
Hi @berendo , sorry that it took so much time. I was busy with other work. Current tcp implementation is only for Linux/Mac where pthreads doesn't need stack attribute, so transport layer doesn't count with it.

Hadatko, 2022-08-01:
Here are some information regarding to init:
[image]