EmbeddedRPC/erpc #280: Add to server option to free codec while real API called

amgross, 2022-06-03:
As far as I understand, while real API called in the server the codec is not in use and all whatever in it won't be used.
So what do you think about adding option (under compilation flags) to release the codec before the API called and create it back after the API finishes?
For systems with little RAM this may mean lots.
We can see that the codec is not really in use while the API because we run codec->reset after that

Hadatko, 2022-06-15:
Hm currently it make sense. Later we wan't option to do not realocate data so user will get data pointing to received message.

amgross, 2022-06-15:
> Later we wan't option to do not realocate data so user will get data pointing to received message.
OK, got your point. You plan it to be all the time or just under specific compilation flags?