EmbeddedRPC/erpc #281: [BUG] Cursor class is inside message buffer class

amgross, 2022-06-06:
**Describe the bug**

1Cursor class is in side message buffer class, although the message buffer don't use its inner cursor and the cursor is using message buffer that it gets as input instead its own message buffer. It makes things unclear and other bugs (In other issues I will open)

**To Reproduce**

**Expected behavior**

there are three options (from better to worth in my opinion):
1. instead of cursor class, to have in the message buffer variable of m_cursor_pos (That is what I suggesting)
2. Have the cursor class be separated
3. Cursor class will use the message class it is inside of.

**Screenshots**

**Desktop (please complete the following information):**

- OS:
- eRPC Version:v1.9.0

**Steps you didn't forgot to do**

- [X] I checked if there is no related issue opened/closed.
- [X] I checked that there doesn't exist opened PR which is solving this issue.

**Additional context**

amgross, 2022-06-08:
@MichalPrincNXP @Hadatko any thoughts?

Hadatko, 2022-06-15:
HI @amgross thank you for all your inputs. I think that what you proposed here https://github.com/EmbeddedRPC/erpc/issues/282 will work nice. We can remove m_remainig and use MessageBuffer variables which can be accessed already like getLength() for write and getUsed for read. 

Here it is your 3. bullet. I think i like it a bit more than 1. bullet.

amgross, 2022-06-15:
OK, I can live with bullet 3.
I prefer instead of `uint8_t *m_pos;` to have `uint16_t m_offset;`, I think it is better to have just one pointer to the buffer and everyone will take offset from it, what do you think?
Also if we want to keep the nested classes, we can use `m_buffer->m_used` instead of `m_buffer->getUsed()`

Hadatko, 2023-09-27:
https://github.com/EmbeddedRPC/erpc/pull/378 ;)

closed 2023-10-24 by MichalPrincNXP