EmbeddedRPC/erpc #282: [BUG] Cursor read may read unstructured data

amgross, 2022-06-06:
**Describe the bug**

Cursore read code is checking that not read outside the remaining:
https://github.com/EmbeddedRPC/erpc/blob/932a8aa2ef603ca52a4c37d62bb3f59a60e17c05/erpc_c/infra/erpc_message_buffer.cpp#L120-L123
But the remaining include unstructured data that is not in used, as it defined as:
https://github.com/EmbeddedRPC/erpc/blob/932a8aa2ef603ca52a4c37d62bb3f59a60e17c05/erpc_c/infra/erpc_message_buffer.cpp#L105
And not as:
https://github.com/EmbeddedRPC/erpc/blob/932a8aa2ef603ca52a4c37d62bb3f59a60e17c05/erpc_c/infra/erpc_message_buffer.h#L357

**To Reproduce**

**Expected behavior**

I think the right way is to remove the m_remaining variable, and use just the buffer pointer and m_used for boundaries. That way also each time that changing m_used the cursor won't read outside. It may be easy as part of #281 

**Screenshots**

**Desktop (please complete the following information):**

- OS:
- eRPC Version: v1.9.0

**Steps you didn't forgot to do**

- [X] I checked if there is no related issue opened/closed.
- [X] I checked that there doesn't exist opened PR which is solving this issue.

**Additional context**

amgross, 2022-06-14:
I prefer to fix it as part of #281 but I can't fix #281 by my own, so I will provide patch here if no one will raise suggestions there

Hadatko, 2022-06-15:
I see the point here. Currently it could be solved with some assert where there could be comparision between cursor remaining and message used. I will look on you other suggestions. Remove m_remaining with m_used is good idea.

amgross, 2022-06-15:
I don't like to add assert in flow that we know can happen in the field

closed 2022-07-26 by MichalPrincNXP