EmbeddedRPC/erpc #284: [FEATURE] Read and write the message just by message buffer memory space

amgross, 2022-06-06:
**Is your feature request related to a problem? Please describe.**

Currently there is difference between the message header and the message body, while the message header is read/write from the stack:
https://github.com/EmbeddedRPC/erpc/blob/932a8aa2ef603ca52a4c37d62bb3f59a60e17c05/erpc_c/infra/erpc_framed_transport.cpp#L54
https://github.com/EmbeddedRPC/erpc/blob/932a8aa2ef603ca52a4c37d62bb3f59a60e17c05/erpc_c/infra/erpc_framed_transport.cpp#L115
the problems with it are:
1. the send happens in two chunks, which is slower
2. In devices that there is difference between memory regions, it may make life harder

**Describe the solution you'd like**

for read:
read first 4 bytes to the message buffer, (and with the curser read from the buffer?)
for write:
In the autogenerated code, after reset the cursor, do dummy write of 4 bytes with the cursor. In FramedTransport::send write to the first 4 bytes the header (if want with the cursor, need to fix first #283 ) and send the whole message in one chunk
**Describe alternatives you've considered**

can do just the write to the buffer in the way I described.
**Steps you didn't forgot to do**

- [X] I checked if there is no related issue opened/closed.
- [X] I checked that there doesn't exist opened PR which is solving this issue.

**Additional context**

Hadatko, 2022-07-12:
This may impact negativelly (forbidden bytes in buffer) transports which doesn't use framedtransport features.

amgross, 2023-09-19:
The PR not really solved this issue. @Hadatko \ @MichalPrincNXP can you please re-open it?

Hadatko, 2023-09-19:
@MichalPrincNXP this is not yet solved. RPMSG tty is not solving this issue. Myabe it was closed automatically so i reopened it.

closed 2023-10-24 by MichalPrincNXP