EmbeddedRPC/erpc #285: [QUESTION] How to generate Python files when IDL files contain C headers?

henrifrancois, 2022-06-10:
**If you didn't find answer in existing open/closed issues you may ask here**

**Steps you didn't forgot to do**

- [X] I checked if there is no related issue opened/closed.
- [X] I checked that there doesn't exist opened/closed PR which is solving this issue.
- [X] I looked in documentation if there is related information.

Hadatko, 2022-06-13:
HI @henrifrancois , could you post erpc file so i can doublecheck its output?

henrifrancois, 2022-06-13:
Hi @Hadatko, I cannot post the files due to their sensitive nature and complexity. I am finding it difficult to reproduce the issue on small files. I will try to update in the next 24 hours on my efforts to reproduce the issue with a minimal example.

henrifrancois, 2022-06-13:
for reference when using erpcgen I receive several instances of "invalid map key: mlComment" as an error.

henrifrancois, 2022-06-14:
As you recommended, changing `@include` annotations to `@c:include` as well as `@external` to `@c:external` seemed to have solved this issue for me. I will mark this as closed.

closed 2022-06-14 by henrifrancois