EmbeddedRPC/erpc #291: PendingClientInfo m_next not updated when aded to active list

bartro, 2022-07-01:
In TransportArbitrator::addPendingClient, when a PendingClientInfo object is removed from a free list that contains multiple objects, its m_next will point to the next object on that free list.
If m_clientLisst is NULL, it will be set to this object but m_next is not updated and will point to an object on the free list.

Issue:
   // Add to active list.
    if (!m_clientList)
    {
        m_clientList = info;
    }
    else
    {
        info->m_next = m_clientList;
        m_clientList = info;
    }

Proposed fix:
    // Add to active list.
    info->m_next = m_clientList;
    m_clientList = info;

Hadatko, 2022-07-01:
Hi @bartro , the fix is already in place: https://github.com/EmbeddedRPC/erpc/pull/199/files

bartro, 2022-07-19:
@Hadatko Thank you, is there a known date for when a release with this fix will be available?

Hadatko, 2022-07-19:
@MichalPrincNXP should be able to answer this question.

MichalPrincNXP, 2022-07-21:
It has been released yesterday, see https://github.com/EmbeddedRPC/erpc/releases

closed 2022-08-01 by Hadatko