EmbeddedRPC/erpc #299: Is ERPC support Windows platform?

PierrePeng, 2022-07-21:
I wonder if ERPC could run on windows? Because I see WIN32 API is used in the project.

thewon86, 2022-08-01:
I need it also.
Today, I build erpc in Cygwin32/64 environment , but it can't be used in Qt5 with MinGW.
When build erpc in Qt5 with MinGW, there are lots errors.

Hadatko, 2022-08-01:
Hi, 
erpcgen can be built under windows in Visual Studio (but we are providing executable files in releases). This application is used to generated shim code for eRPC applications. 
eRPC lib can be likely built only for Linux. You should be likely possible to built app for windows with serial transport. We are not focusing on supporting windows eRPC applications. But you should be able to do it if you implement transport (other than serial) for windows.

thewon86, 2022-08-03:
Thanks,
I did it. it works. now, erpc_tcp_transport can be compiled with MinGW / cygwin GCC / linux GCC at the same time.

huiyeruzhou, 2022-11-05:
> Thanks, I did it. it works. now, erpc_tcp_transport can be compiled with MinGW / cygwin GCC / linux GCC at the same time.

Hi, I'm trying the same as you did and notice that `erpc_tcp_transport.cpp` includes some header files that are not exist in Windows. Should I use winsock to implement it?
By the way, the motivation to compile erpc in windows is that I want use the class designer in VS to generate UML for erpc, maybe there is a better way to do it >_<...

thewon86, 2022-11-06:
> 

https://thewon.gitee.io/2022/0902/erpc-transport-winsock2/

Hadatko, 2023-02-05:
Hi @thewon86 , could you provide PR?

closed 2023-09-19 by MichalPrincNXP