EmbeddedRPC/erpc #303: ERPC support over inter process/thread communication

tmordeko, 2022-08-18:
Hi,

Wondering if there is a plan to add inter process communication for ERPC transports ?

Also, appreciate if you could provide simple example how to init InterThreadBufferTransport between two threads (some pseudo code would help here to demonstrate how to use this flavor)

Thanks

Hadatko, 2022-08-18:
Hello, we have this transport: https://github.com/EmbeddedRPC/erpc/blob/develop/erpc_c/transports/erpc_inter_thread_buffer_transport.hpp

But i think it wasn't used / tested for longer time of period. Maybe it has to be updated.

tmordeko, 2022-08-19:
Thanks @Hadatko I was referring to inter processes communication (between 2 different processes, not threads)

Regarding https://github.com/EmbeddedRPC/erpc/blob/develop/erpc_c/transports/erpc_inter_thread_buffer_transport.hpp do you have an example of usage especially on how to initiate the transport 

Thanks