EmbeddedRPC/erpc #305: [QUESTION] Is eRPC just support 256 functions?

thewon86, 2022-08-24:
Hi All,

I create a interface with more than 300 functions , a request id with 342(0x156) become 86(0x86) in server side.
I find the code `uint32_t header = (kBasicCodecVersion << 24u) | ((service & 0xffu) << 16u) | ((request & 0xffu) << 8u) | ((uint32_t)type & 0xffu);` , that make the request value (0-255)

So, how could i add more than 256 functions? Adding more interfaces(services)? or modify the code upside?

Thanks.

Hadatko, 2022-08-28:
Hi @thewon86 , yes currently it is 256 function types per interface. https://github.com/EmbeddedRPC/erpc/blob/d8f6a20d7ca4379d1f0ea437714a96b135a7057b/erpc_c/infra/erpc_basic_codec.cpp#L32
You would have to split functions into more interfaces, or write own basic codec with supporting more functions. Anyway we should add some asserts to catch this type of errors in debug mode.

thewon86, 2022-09-01:
> 

I see . Thank you.

KcMeterCEC, 2024-12-01:
If i want to support more than 255 function id name in one interface,two places should i change:

1. `erpcgen\src\templates\cpp_interface_header.template`:

```cpp
// static const uint8_t {$getClassFunctionIdName(fn)} = {$fn.id};
static const uint16_t {$getClassFunctionIdName(fn)} = {$fn.id};
```

2. `erpc_c\infra\erpc_basic_codec.cpp`:

```cpp
void BasicCodec::startWriteMessage(message_type_t type, uint32_t service, uint32_t request, uint32_t sequence)
{
    uint64_t header =
        (kBasicCodecVersion << 40u) | ((service & 0xffu) << 32u) | ((request & 0xffffu) << 8u) | ((uint32_t)type & 0xffu);
 //...
}

void BasicCodec::startReadMessage(message_type_t &type, uint32_t &service, uint32_t &request, uint32_t &sequence)
{
    uint32_t header;

    read(header);

    if (((header >> 40) & 0xffU) != kBasicCodecVersion)
    {
        updateStatus(kErpcStatus_InvalidMessageVersion);
    }

    if (isStatusOk())
    {
        service = ((header >> 32 & 0xffU);
        request = ((header >> 8) & 0xffffU);
        type = static_cast<message_type_t>(header & 0xffU);

        read(sequence);
    }
}

```

is it  correct? @Hadatko

amgross, 2024-12-01:
Something here is wrong I think:
```
uint32_t header;
read(header);
```
You need it to be uint64_t.

Need to take in account that the header will now be 64 bits instead of 32, but 24 bits are reserved.
Also, if you need JAVA/PYTHON you will need to fix them as well.
I can't think now on something else that missing, but I didn't test it.

KcMeterCEC, 2024-12-01:
You are right,I need to test these changes. @amgross

amgross, 2024-12-01:
@KcMeterCEC another option is to use 12 bits for the request and 4 for the service, this way the header stays 32 bits.
Just suggesting, whatever better for your use case

thewon86, 2024-12-03:
> @KcMeterCEC another option is to use 12 bits for the request and 4 for the service, this way the header stays 32 bits. Just suggesting, whatever better for your use case

right, it's the some i had done