EmbeddedRPC/erpc #310: Linux - FreeRTOS communication using callbacks

anakorenko, 2022-09-12:
Hello,  I'm trying to implement the next erpc client-server communication. My erpc server is running under Cortex-M4 microprocessor under FreeRTOS and CMSIS_UART as communication interface.
My client is running under Linux. Client-server communication is working fine, but I got some difficulties in setting server callback back to client.
In all discussions I've found all server to client callbacks are working when both client and server using the same architecture: e.g server and client are both under Linux and using pthreads, or server and client are using FreeRTOS threads.
Is it possible to implement server callbacks to client when they are running on different archs?

Hadatko, 2022-09-12:
Hi @anakorenko , different architecture shouldn't be an issue. But currently it is supported in C only. The way how it works is that erpc is sending index to an array of functions which is created on both sides by erpc.

anakorenko, 2022-09-12:
Thank you for quick replay. If architecture isn`t an issue how to configure threading in erpc_config.h? Can I setup client for pthread and server for freertos multithreading?

Hadatko, 2022-09-13:
Hi @anakorenko , as you will have two application, each application will have its own erpc_config.h file where you can specify different enviroment settings.

anakorenko, 2022-09-13:
Thank you, your answer helped me. But i have another questions do i need to use arbitrator in this case?

Hadatko, 2022-09-13:
You are welcome. Arbitrator need be used if you have client and server on each side. So you want do client requests from both sides and servers will respond on both sides.

anakorenko, 2022-09-13:
Greate, thanks, now i`m trying to make my callbacks work using examples from test directory, but somehow receiving strange error when trying to compile server-side of the application:
```
/usr/local/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: _build/nrf52840_xxaa/erpc_demo_server.cpp.o: in function `linuxTonordic_service::myFun_shim(erpc::Codec*, erpc::MessageBufferFactory*, unsigned long)':
/home/artem/Documents/Packetcraft/ProjFreeRtos/pca10056/blank/armgcc/../../../erpc_demo_server.cpp:76: undefined reference to `erpc_malloc'
/usr/local/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/artem/Documents/Packetcraft/ProjFreeRtos/pca10056/blank/armgcc/../../../erpc_demo_server.cpp:111: undefined reference to `erpc_free'
/usr/local/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/artem/Documents/Packetcraft/ProjFreeRtos/pca10056/blank/armgcc/../../../erpc_demo_server.cpp:111: undefined reference to `erpc_free'
/usr/local/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /home/artem/Documents/Packetcraft/ProjFreeRtos/pca10056/blank/armgcc/../../../erpc_demo_server.cpp:111: undefined reference to `erpc_free'
```
This errors occurs only if i use `out` in my function arguments

Hadatko, 2022-09-13:
You need choose erpc_port_*.cpp file from https://github.com/EmbeddedRPC/erpc/tree/develop/erpc_c/port

anakorenko, 2022-09-13:
Yep, now it works, and what about second question?

Hadatko, 2022-09-13:
Sorry i missed your question. No, you need also pass onOff as out parameter for function (same call or new call)
e.g:
myFun(callback, &pCallback1_out, &onOff);
of
myFun2(out onOFF)

anakorenko, 2022-09-16:
Hello again, can i use arbitrator on linux - rtos communication setup? Thanks

Second222None, 2022-12-01:
Hi, I am also doing a similar thing. The difference is that the communication middleware is USB_CDC. 
I focus on the server-side application now. And FreeRTOS runs on STM32.
According to the [Porting Guide](https://github.com/EmbeddedRPC/erpc/wiki/Porting-Guide), I choose the `erpc_port_freertos.*`. Here is the episode from my CMakeList.txt:

```
file(GLOB_RECURSE SOURCES "startup/*.*" "Middlewares/*.*" "Drivers/*.*" "USB_DEVICE/*.*" "Core/*.*" 
"Communication/*.*"
"autogen/*.*" 
"erpc/erpc_c/config/*.*"
"erpc/erpc_c/setup/erpc_setup_usb_cdc.*"
"erpc/erpc_c/setup/erpc_server_setup.*"
"erpc/erpc_c/setup/erpc_client_setup.*"
"erpc/erpc_c/transport/erpc_usb_cdc_transport.*"
"erpc/erpc_c/transport/erpc_serial_transport.*"
"erpc/erpc_c/port/erpc_config_internal.h"
"erpc/erpc_c/port/erpc_endianness_agnostic_example.h"
"erpc/erpc_c/port/erpc_endianness_undefined.h"
"erpc/erpc_c/port/erpc_port_freertos.cpp"
"erpc/erpc_c/port/erpc_port.h"
"erpc/erpc_c/port/erpc_setup_extensions_freertos.cpp"
"erpc/erpc_c/port/erpc_setup_extensions.h"
"erpc/erpc_c/port/erpc_threading_freertos.cpp"
"erpc/erpc_c/port/erpc_threading.h"
)
```

I received a compiling error:

```
In file included from /home/birdman/FOC/erpc/erpc_c/setup/erpc_setup_usb_cdc.cpp:12:
/home/birdman/FOC/erpc/erpc_c/transports/erpc_usb_cdc_transport.hpp:21:10: fatal error: fsl_component_serial_manager.h: No such file or directory
   21 | #include "fsl_component_serial_manager.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/FOC.elf.dir/build.make:830: CMakeFiles/FOC.elf.dir/erpc/erpc_c/setup/erpc_setup_usb_cdc.cpp.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/FOC.elf.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

What am I missing? Does modification of `erpc_erpc_usb_cdc_transport.*` required?
I am new to this. Can anyone help?
Thank you.

Hadatko, 2022-12-04:
@anakorenko Arbirator should be possible for different OS.

Hadatko, 2022-12-04:
@Second222None based on your output you are missing Linux or NXP SDK dependencies. It means there is no port for STM32 and you should do it on your own if you want it. If you provide PR in that case we will appretiate it.