EmbeddedRPC/erpc #311: erpcgen from release 1.9.1 generates an incorrect server file

xanderhendriks, 2022-09-15:
**Describe the bug**
Using erpcgen from release 1.9.1 generates a server file with the inclusion of erpc_server.hpp instead of erpc_server.h

**To Reproduce**
run erpcgen on an erpc IDL file

**Expected behavior**
The <project>_server.h file includes the erpc_server.h that can be found in the release 1.9.1 tagged files and not the renamed erpc_server.hpp which is only there in the develop branch.

**Screenshots**

**Desktop (please complete the following information):**

- OS: Windows
- eRPC Version: 1.9.1

**Steps you didn't forgot to do**

- [x] I checked if there is no related issue opened/closed.
- [x] I checked that there doesn't exist opened PR which is solving this issue.

**Additional context**

Hadatko, 2022-09-19:
Hi @MichalPrincNXP, it looks like we need regenerate all binaries for release. It looks like they are did from different commit than release code version.

I tested manual build of erpcgen on master and output looks ok. But when i downloaded Linux version then i got #include "erpc_manually_constructed.hpp" instead of #include "erpc_manually_constructed.h".
This is different issue than above, but similar.

xanderhendriks, 2022-09-20:
Thanks for picking this up guys. For now I'm just using the 1.9.0 release.

MichalPrincNXP, 2022-09-21:
Hello, thanks for reporting that, I have probably uploaded different binaries, I am sorry ... attached are the v1.9.1 one, could you please have a try and confirm all is ok before I upload them to v1.9.1 release artifacts? Thanks a lot.
[erpcgen_1_9_1.zip](https://github.com/EmbeddedRPC/erpc/files/9615133/erpcgen_1_9_1.zip)

Hadatko, 2022-09-21:
Under Linux it looks fine. @xanderhendriks could you try this binary?

MichalPrincNXP, 2022-09-23:
Binaries updated in https://github.com/EmbeddedRPC/erpc/releases/tag/1.9.1

xanderhendriks, 2022-09-26:
Thanks a lot for fixing this. I can confirm it's now working properly on my development laptop under windows and our Github action under linux.

Hadatko, 2022-10-11:
> The windows version is also okay. Please let me know when the files are uploaded, so I can configure my CI to grab the correct files.
> […](#)

Hi @xanderhendriks , as @MichalPrincNXP closed the issue already and mentioned that binaries are up to date, i expect that everything is correct now.

xanderhendriks, 2022-10-11:
> > The windows version is also okay. Please let me know when the files are uploaded, so I can configure my CI to grab the correct files.
> > […](#)
> 
> Hi @xanderhendriks , as @MichalPrincNXP closed the issue already and mentioned that binaries are up to date, i expect that everything is correct now.

Sorry, I'm not sure what happened there. I had written that comment **before** the previous comment 15 days ago where I wrote that all was OK and thanked you guys for the quick response.

So please ignore this message. I have confirmed that all is working as expected.

Hadatko, 2022-10-11:
Perfect ;) No problem ;)