EmbeddedRPC/erpc #313: Erpc callbacks

Kindzer, 2022-09-19:
Hello @Hadatko, I've started to research eRPC for similar task as in #310  and I'm going through similar difficulty (it's about that I cannot call callback from server side to client and execute it there). So I tried to use your examples to find answer.

1. First I found question #101 and also ran https://github.com/EmbeddedRPC/erpc/tree/develop/test/test_callbacks. All of those tests init callbacks from client to server and as I understand the callback itself executes on server. In tests it was also shown that we can return the pointer of that callback back to client but that is not what I exactly need.

2. In above topic it was also said that there is a need to use transport arbitrator for resolving that issue. So I tried to work with that example and to add a callback functionality to arbitrator example. I can call nested functions now and so on but I still cannot run the callback on server but execute on client.

On client I'm using just on call like _InitCallback(myCallback);_. When definition of _myCallback_ is on server side, everything works just fine, but the callback is executed in server side. But when the definition is on client side I cannot build the example because of multiple declarations on _myCallback_.

Maybe I am missing something? Can you please explain how registered callback on server side from client can be called back to client and be executed there?

Kindzer, 2022-09-19:
UPD:
Looking on example which erpc gives, it seems to me that with callbacks the server and the client must have same memory space for the callback correct work (multi-core examples, or unittest examples). Is callback functionality is suitable when server (e.g Linux PC) and client (e.g STM32) have separate memory space?

Hadatko, 2022-09-21:
Hi @Kindzer , callback should work on different platforms in C language. Python is not supported now. We are serializing callback functions as indexes in our array of functions.
In test if you look on callback2 function, you will find that this one has definition on client and called from server. For this case both side actually are servers and clients. You need use arbitrator to be able to do this hybrid model. 

You need two interfaces. On one side you will link firstInterfaceClient with secondInterfaceServer. On other side secondInterfaceClient with firstInterfaceServer.

Kindzer, 2022-09-21:
Hello @Hadatko. Your answer help a lot. I tried to make callbacks as you said with arbitrator based on test_arbitrator example and everything works as I want.

Probably the last question which I want to clarify is about setting up the correct erpc_config.h. So I have client(STM32 under FreeRTOS) and server (PC under Linux), so I need to choose the correct threading model.

From erpc_config.h I'm getting next information:

_//! @def ERPC_THREADS
//!
//! @brief Select threading model.
//!
//! Set to one of the @c ERPC_THREADS_x macros to specify the threading model used by eRPC.
//!
//! Leave commented out to attempt to auto-detect. Auto-detection works well for pthreads.
//! FreeRTOS can be detected when building with compilers that support __has_include().
//! Otherwise, the default is no threading._

From this message as I understand I can leave this macro commented and erpc will auto-detect FreeRTOS on STM32 and pthreads on Linux by itself. (I'm compiling with gcc arm compilers so __has_include() must be supported)

Am I correct?

Hadatko, 2022-09-21:
Hi @Kindzer , nice progress. 
Details about autodetection can be found in this file https://github.com/EmbeddedRPC/erpc/blob/develop/erpc_c/port/erpc_config_internal.h
So you can keep them commented. But if you would face some issue, you can have different config for linux and for stm.

Kindzer, 2022-09-21:
Thanks a lot, found out how to set different config files separately to server and to the client.

Thanks a lot for the help again.

closed 2022-12-14 by Hadatko