EmbeddedRPC/erpc #314: [BUG]  Issue while creating windows erpcgen exe

amgross, 2022-09-21:
**Describe the bug**
Trying create my own version of Windows erpcgen, I got error. Running the failure command with command line the error was:
erpcgen_parser.y:22.9-19: error: %define variable 'parse.error' is not used

Removing this line from erpcgen_parser.y resolved the problem and I succeeded to compile my own erpcgen.

I am using visual studio 2017 pro . in the beginning I got some issue from visual studio that my windows SDK is not compatible with the one of the visual studio project, so I retargeted the project to 10.0.19041.0 SDK
**To Reproduce**

**Expected behavior**

**Screenshots**

**Desktop (please complete the following information):**

- OS:
- eRPC Version:
v1.8.1
**Steps you didn't forgot to do**

- [x] I checked if there is no related issue opened/closed.
- [X] I checked that there doesn't exist opened PR which is solving this issue.

**Additional context**

Hadatko, 2023-01-04:
Hi @amgross , this issue can be closed or still relevant and you closed just PR?

amgross, 2023-01-08:
Hi, can be closed, unless someone more will face this issue

closed 2023-01-08 by amgross