EmbeddedRPC/erpc #315: Structures definition on server side

anakorenko, 2022-09-22:
Hello erpc team, as far as i understand, there is no need to declare struct on server side or client. Moreover if i define struct on server side i receive errors:
`error: redefinition of ‘struct nested_t’`
`error: redefinition of ‘struct elements_t’`
My structs defined on server side as follows:
`typedef struct elements_t elements_t;
struct elements_t
{
    int8_t elem;
    int8_t elem2;
    int8_t *word;
};
 
typedef struct nested_t nested_t;

struct nested_t
{
    elements_t struct1;
    elements_t struct2;
};`

and in erpc file:
`struct elements_t
{
    int8 elem
    int8 elem2
    byref int8 word
}

struct nested_t
{
    elements_t struct1
    elements_t struct2
}`
Is there any way to keep declaration of structure on server side? Thank you

Hadatko, 2022-09-22:
Hello, if you don't want erpc to generate declarations, use @external annotation ;)

anakorenko, 2022-09-22:
Thanks, it works. And i have another question. I have structure defined on server (not generated by erpc) an i want to call it on client using callback.

Hadatko, 2022-12-14:
Hi @anakorenko , if you don't want use that structure on client side, then maybe include file in erpc IDL file where for client the file will contain structure definition with some body (not important implementation). In server the include file will be empty